openv/openv #602: ESPHome Optolink daten per HTTP senden

Jumbo125, 2023-08-01:
Hallo zusammen

weiß jemand wie man die Daten, welche man erhält auch mittels http weitersenden kann?
Vielen Dank!

j0ta29, 2023-08-02:
Hallo @Jumbo125 , ESPHome sendet seine Daten aktiv über die HomeAssistant-API, die auf REST basiert, das wiederum auf HTTP aufsetzt. Theoretisch könntest Du die API-Aufrufe auch an eine eigene kompatible Schnittstelle schicken (nicht Home Assistant), ich vermute nur, dass das nicht ganz trivial ist. 
Ich weiß nicht ob das Deine Frage beantwortet. Eventuell müsstest Du nochmal detaillierter erläutern, welchen Anwendungsfall Du mit "mittels http weitersenden" im Kopf hast.

Jumbo125, 2023-08-02:
Vielen Dank für die Hilfe. Ich müsste die Daten via http an mei  script senden.
Z.b  
10.0.0.140/meinphpscript.php?boilertemperatur=wert

j0ta29, 2023-08-02:
Ich denke, hier sollte Dir das helfen: https://esphome.io/components/http_request.html

Jumbo125, 2023-08-02:
> Ich denke, hier sollte Dir das helfen: https://esphome.io/components/http_request.html

danke dir. das habe ich schon gefunden, aber leider bin ich mir mit der umsetzung nicht ganz sicher. esphome ist neu für mich. yaml leider auch.... wie kann ich den die parameter von der LWP in eine variable schreiben um später diese in mein http-request einzubinden??
hier mal mein bspw code:
`esphome:
  name: optolink

esp8266:
  board: d1_mini

# Enable logging
logger:

external_components:
  - source: github://pr#4453
    components: [ optolink ]

wifi:
  ssid: "xxxxxxxxxxxxxxxxxxxxx"
  password: "xxxxxxxxxxxx!"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Optolink Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxx"

captive_portal:

# optolink parameter
optolink:
  protocol: P300             # P300 oder KW
  logger: enable             # wenn später alles rund läuft, wieder entfernen 
  state: Status              # wenn später alles rund läuft, wieder entfernen
  device_info: Gerätekennung # dient zur Identifizierung der Vitotronic 

logger:
  hardware_uart: UART1
  level: DEBUG

sensor:
  - platform: optolink
    name: ${name} Kesseltemperatur
    address: 0x0810
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature

number:
  - platform: optolink
    name: ${name} Sollwert Raumtemperatur
    unit_of_measurement: °C
    address: 0x2306
    bytes: 1
    min_value: 3
    max_value: 37
    step: 1
    mode: box
    icon: "mdi:home-thermometer"
    device_class: temperature

#http request
http_request:
  useragent: esphome/device
  timeout: 10s
  esp8266_disable_ssl_support: false
  follow_redirects: true

on_...:
  - http_request.post:
      url: https://esphome.io
      verify_ssl: false
      json:
        Kesseltemperatur: kesseltemperatur variable
        Sollwert_Raumtemperatur: Raumtemperatur variable

    `

j0ta29, 2023-08-02:
Innerhalb der sensor Definition solltest Du einen on_value-Eintrag einfügen, wo du in einem Lambda-Ausdruck dann per x auf den Sensorwert zugreifen kannst. Ein Beispiel findest Du unter https://esphome.io/components/sensor/index.html#on-value

Jumbo125, 2023-08-03:
Danke dir. wirkt sehr umfangriech :-) 
ich überlege zurzeit, obi ich nicht https://github.com/dumpfheimer/OptoProxy verwende, das sieht genau passend aus. aber wird dieses noch unterstütz oder ist das project abgeschlossen???
das gleiche frage ich mich auch über https://github.com/bertmelis/VitoWifi
dies sieht auch sehr gut aus, jedoch schaut das anlegen von datenpunkten sehr kompliziert aus....

j0ta29, 2023-08-03:
Zu Optoproxy kann ich leider nichts sagen. 
Vit0Wifi kann ich empfehlen, weil ESPHome-Optolink-Komponente diese Bibliothek auch verwendet ;-)
Hat zwar lange kein Update mehr erhalten, aber läuft sehr meiner Erfahrung nach stabil.
Allerdings wirst Du auch hier das Problem haben, dass Du den HTTP-Request in C++ selbst programmieren musst. Der ganze Komfort von ESPHome (u.a. OTA, Web-Interface ....) geht Dir dann halt verloren.

Jumbo125, 2023-08-03:
Danke dir
C++ wäre kein Problem. dafür gibt es auch schon eigene http request libarys. Die Yaml Datei ist leider völliges neuland für mich.