openv/openv #605: ESPHome ESP32 Homeassistant -> kein Log

andyyyy2306, 2023-08-09:
Hallo !
Ich versuche nach dem Wechsel von Domoticz mit vcontrold zu Homeassistant mit [ESPHome Optolink](https://github.com/openv/openv/wiki/ESPHome-Optolink) meine Wärmepumpe (WärmePumpe 222-G | CD70 | Gerätecode 2000 44 | KW2) auszulesen.
Optolink Adapter habe ich von [ESPHome Optolink](https://github.com/openv/openv/wiki/ESPHome-Optolink)  nachgebaut

Leider ohne Erfolg. :-(
Anbei meine YAML und das Log. Im Log sehe ich in den leeren Zeilen nur Kurz einen Text der dann verschwindet.

      

[viessmann32.yaml](https://github.com/openv/openv/files/12300539/viessmann32.yaml.txt)

[image]

Falls Jemand eine Idee hat wo ich ansetzen kann oder zumindest das Log sichtbar zu machen wäre ich dankbar !

LG, Andyyyy

j0ta29, 2023-08-09:
Hallo @andyyyy2306 ,

die Komponenten `optolink`verwendet (aktuell noch) fest verdahtet UART0. Das gleiche macht die Komponente `logger` - falls man sie nicht anders konfiguriert. Deshalb solltest Du das noch mit in Deine Konfiguration aufnehmen:
```
logger:
  hardware_uart: UART1
```
Siehe auch den Hinweis unter https://deploy-preview-2737--esphome.netlify.app/components/optolink

Oder das Logging auf UARTs ganz ausschalten:
```
logger:
  baud_rate: 0
```

Wenn Du die Log-Ausgabe auch ohne `esphome` Kommando sehen willst, hilft auch die `web-server`Komponente:
```
web_server:
  port: 80
```

Wenn das auch nicht hilft, müsstest Du prinzipiell mal die Funktionsfähigkeit deines DIY-Adapters überprüfen:
Der Anwender _derwassi_ hat dazu unter den folgenden beiden Links beschrieben, wie der das macht. Sender und Empfänger gegeneinander halten (ganz ohne Viessmann) und prüfen, ob der Sender den Empfänger triggert.
Hier die Links dazu:

https://github.com/esphome/esphome/pull/4453#issuecomment-1546306287
https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702

Viel Erfolg!

andyyyy2306, 2023-08-10:
Hallo @j0ta29 
Danke für die Hinweise.
Ich habe es nun mit ESP8266 und ESP32 versucht.
Komisch ist das ich weder Logeinträge von Vitowifi in der ESPHome Konsole sehe, noch der Webserver anläuft.
Hab es schon mit Wemos D1 mini pro, NodeMCU etc. versucht. 
Anfragen von Vitowifi müsste ich im Log doch sehen, auch ohne Hardware dahinter oder ?

```
esphome:
  name: viessmann
  friendly_name: Viessmann

esp8266:
  board: d1_mini_pro

logger:
    hardware_uart: UART1
    baud_rate: 0

web_server:
  port: 80

api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.1.198
    gateway: 192.168.1.254
    subnet: 255.255.255.0

    
external_components:
  - source: github://pr#4453
    components: [ optolink ]

optolink:
  protocol: KW            # P300 oder KW
  device_info: Device Info
  logger: enable

sensor:
  - platform: optolink
    name: Aussentemperatur
    address: 0x0800
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature

```

j0ta29, 2023-08-11:
@andyyyy2306 , Du solltest in der `logger` Komponente noch den Level auf `DEBUG` setzen.
```
logger:
  hardware_uart: UART1
  baud_rate: 0
  level: DEBUG
```
Danach müsstest Du im Web-Interface des ESP eine Ausgabe vergleichbar zu dieser hier sehen:
[image]

Im Web-Inferface von ESPHome fehlen leider die Ausgaben von `VitoWiFi`. Dort werden sie nur als leere Zeilen dargestellt.

Welche Probleme hast Du denn mit der `web_server` Kompoente?

andyyyy2306, 2023-08-12:
@j0ta29 
Ich denke die web_server Komponente hatte ich an der falschen Stelle, die funktioniert nun.
DAs Loglevel hab ich auch auf Debug gesetzt.
Aber für mich sieht es irgendwie aus als würde die optolink bzw. Vitowifi Komonente nie aktiviert.
Mein Log sieht so aus
[image]

j0ta29, 2023-08-12:
@andyyyy2306 , die Ausgabe "read value of datapoint..." sagt eigentlich nur, dass der Wert von der Therme angefordert wird. Sollte also eher "requesting value..." heißen.
Die Ausgabe, die Du bekommst ist typisch für einen noch nicht funktionierenden Adapter bzw. eine nicht korrektes Zusammenspiel mit der Therme.
Ich würde Dir empfehlen, den Adapter mal mit _derwassis_ s Methode zu testen (siehe oben):
https://github.com/esphome/esphome/pull/4453#issuecomment-1546306287
https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702

andyyyy2306, 2023-08-14:
Hallo  @j0ta29 ,
danke für die Hilfe.
Ich konnte mit einem Wemos D1 mini pro (ESP8266) den Optolink,wie hier beschrieben https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702, testen. jedoch nur auf GPIO4 und GPIO5. Wenn ich GPIO1 und GPIO3 verwendet habe ging es nicht.
Dann hab ich versucht auf ESP32 umzuschwenken um die rx und tx Pins vergeben zu können.
Hat dann auch funktioniert und ich habe mir wieder die config für Optolink eingefügt.

```
esphome:
  name: viessmann32
  friendly_name: Viessmann32

esp32:
  board: esp32dev
  framework:
    type: arduino
  
logger:
  hardware_uart: UART1
  level: DEBUG

external_components:
  - source: github://pr#4453
    components: [ optolink ]

optolink:
  protocol: KW
  device_info: Viessmann
  logger: enable
  rx_pin: 22
  tx_pin: 23

dallas:
  - pin: 2

web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.1.194
    gateway: 192.168.1.254
    subnet: 255.255.255.0

sensor:
  
  - platform: dallas
    address: 0xdf01144a1611aa28
    name: "Sole Vorlauftemperatur"

  - platform: dallas
    address: 0xf601144a0671aa28
    name: "Sole Rücklauftemperatur"
 
 - platform: optolink
    name: Aussentemperatur
    address: 0x0800
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature
```
  

Nun sehe ich folgende Logausgabe, jedoch nur mit Timeout´s.
[image]

devganny, 2023-08-14:
@andyyyy2306: Bist Du hier weitergekommen? Ich versuche auch vergebens mit dem Wemos D1 mini pro weiterzukommen. Ich sehe auch nur die read value Meldungen. Habe jetzt alles schon 10x geprüft und komme einfach nicht weiter.

j0ta29, 2023-08-14:
Hallo @andyyyy2306 , 
Die Fehlermeldung "timeout" sagt eigentlich nur aus, dass nach dem Absenden der Werteanforderung über die Sendediode im vorgesehenen Zeitfenster keine Antwort am Fototransistor ankam. Wieso das so ist, hmmmm .... gute Frage.
Auf der Seite https://github.com/openv/openv/wiki/Ger%C3%A4te finde ich leider keine Angabe zum Protokoll für dein Gerät. Ein bisschen wundert mich, dass Du das Protokoll "KW" eingetragen hast. Ich dachte, dass das eher für ältere Modelle ist. Hast Du schon mal einen Versuch mit dem Protokoll "P300" unternommen?
Ich würde aber in der Konfiguration der `logger` Komponente sicherheitshalber erst mal bei `baud_rate` 0 bleiben. Du fragst die Debug-Ausgabe ja nicht am seriellen Port ab. Nicht dass es doch noch eine Kollision mit der seriellen Schnittstelle des Optolink-Adapters gibt.

j0ta29, 2023-08-14:
@devganny , der erste Funktionstest des DIY-Adapter ist zu prüfen, ob die Photodiode überhaupt sendet. Das geht in der Regel mit jedem Smartphone. Durch die Kamera des Handys machst du die IR-Signale der Diode für das menschliche Auge sichtbar. Wenn dort was blinkt, aber immer noch nichts vom Adapter empfangen wird, hilft der Test _von derwassis_ :
https://github.com/esphome/esphome/pull/4453#issuecomment-1546306287
https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702

andyyyy2306, 2023-08-14:
@j0ta29 
KW deshalb weil ich Jahrelang mit einem Rpi und vcontrold meine Heizung ausgelesen habe.
In meinen alten Configs wird dieser Abschnitt für die Verbindung verwendet.

```
  <protocols>
		<protocol name='KW2'>
			<macros>
				<macro name='SYNC'>
					<command>SEND 04;WAIT 05</command>
				</macro>
				<macro name='GETADDR'>
					<command>SEND 01 F7</command>
				</macro>
				<macro name='SETADDR'>
					<command>SEND 01 F4</command>
				</macro>
			</macros>
			<commands>
				<command name="getaddr">
					<retry>3</retry>
					<recvTimeout>4000</recvTimeout>
					<send>SYNC;GETADDR $addr $hexlen;RECV $len $unit</send>
				</command>
				<command name="setaddr">
					<send>SYNC;SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
				</command>
			</commands>
		</protocol> 
  </protocols>

```
sieht Vitowifi anders aus ?

devganny, 2023-08-14:
Hat es jemand mit dem D1 Mini Pro bisher zum laufen gebracht oder gibt es da generell Probleme?

devganny, 2023-08-14:
> @devganny , der erste Funktionstest des DIY-Adapter ist zu prüfen, ob die Photodiode überhaupt sendet. Das geht in der Regel mit jedem Smartphone. Durch die Kamera des Handys machst du die IR-Signale der Diode für das menschliche Auge sichtbar. Wenn dort was blinkt, aber immer noch nichts vom Adapter empfangen wird, hilft der Test _von derwassis_ : [esphome/esphome#4453 (comment)](https://github.com/esphome/esphome/pull/4453#issuecomment-1546306287) [esphome/esphome#4453 (comment)](https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702)

Geht bei mir z.B. nicht über die Standard RX/TX Pins alternative PINS muss ich noch testen, aber das Modul kann ja wohl nur mit den Standard Ports auf den W1 laufen....

andyyyy2306, 2023-08-14:
Danke für die Hilfe @j0ta29 !!
Habe es nun geschafft mal die Aussentemperatur zu lesen, jetzt muss ich noch die anderen Datenpunkte eintragen.
Der Fehler lag nun an einer kalten Lötstelle meines Optolink Adapters. 

Meine funktionierendes Setup ist nun ein ESP32 mit dieser Bauanleitung [https://github.com/openv/openv/wiki/ESPHome-Optolink](url) und folgendem Code:

```
esphome:
  name: viessmann32
  friendly_name: Viessmann32

esp32:
  board: esp32dev
  framework:
    type: arduino
  
logger:
  hardware_uart: UART1
  level: DEBUG
  baud_rate: 0   

external_components:
  - source: github://pr#4453
    components: [ optolink ]

optolink:
  protocol: KW
  device_info: Viessmann
  logger: enable
  rx_pin: 22
  tx_pin: 23

web_server:
  port: 80

# Enable Home Assistant API
api:
  encryption:
    key: " "

ota:
  password: " "

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.1.194
    gateway: 192.168.1.254
    subnet: 255.255.255.0

# Example configuration entry
dallas:
  - pin: 2

sensor:
  
  - platform: dallas
    address: 0xdf01144a1611aa28
    name: "Sole Vorlauftemperatur"

  - platform: dallas
    address: 0xf601144a0671aa28
    name: "Sole Rücklauftemperatur"

  - platform: optolink
    name: Aussentemperatur
    address: 0x0800
    bytes: 2
    div_ratio: 10
    accuracy_decimals: 1
    unit_of_measurement: °C
    device_class: temperature
  

```

[image]

j0ta29, 2023-08-14:
@andyyyy2306 , das freut mich, dass Du nach langem Kampf Erfolg hattest.

j0ta29, 2023-08-14:
> Geht bei mir z.B. nicht über die Standard RX/TX Pins alternative PINS muss ich noch testen, aber das Modul kann ja wohl nur mit den Standard Ports auf den W1 laufen....

@devganny , soweit ich den D1 mini (bzw. den ESP8266) verstanden habe, ist die Hardware-unterstützte serielle Schnittstelle hart verdrahtet mit GPIO1 und GPIO3. Die anderen PINs sind meines Wissens nur über einen Software-UART ansteuerbar, der wohl aber nicht so stabil arbeitet wie der Hardware-UART. 
Dass der D1 mini pro mit der `optolink` Komponente arbeitet, kann ich bestätigen. Das ganze läuft bei mir stabil seit mehr als einem halben Jahr.

devganny, 2023-08-14:
> @devganny , soweit ich den D1 mini (bzw. den ESP8266) verstanden habe, ist die Hardware-unterstützte serielle Schnittstelle hart verdrahtet mit GPIO1 und GPIO3. Die anderen PINs sind meines Wissens [...]

Ich habe gerade, nur um zu testen ob die Komponenten an sich funktionieren das ganze mal auf Pin 04 und 05 gemäß _derwassis_ getestet und das klappt. Sobald ich aber wieder auf RX und TX umziehe funktioniert weder der Test senden empfangen über Pin 1 und 3, noch das Abfragen der Werte der Vitodens 200. Bin mit meinem Latein im Ende. 
Verwende das **AZDelivery D1 Mini V3 NodeMCU ESP8266EX**

andyyyy2306, 2023-08-15:
@devganny deshalb habe ich auf ESP32 gewechselt.

devganny, 2023-08-15:
> @devganny deshalb habe ich auf ESP32 gewechselt.

ich werde jetzt mal noch einen älteren D1 ausprobieren, ansonsten wird mir wohl auch nichts anderes übrigbleiben als auf esp32 zu wechseln. Ist nur ärgerlich, da das D1 schön klein ist und ich dafür jetzt schon ein passendes Gehäuse designt und gedruckt habe das perfekt in das "V" passt. Aber gut, dann nochmal zurück auf "Los" :-)

devganny, 2023-08-15:
So ich habe jetzt eine älters D1 Modell getestet in der V2, damit klappt alles. Das Problem sind offenbar die V3 Boards, warum auch immer. Nur als Warnung für alles anderen, bevor sie hier wie ich sinnlos Stunden mit Fehlersuche vergeuden.

andyyyy2306, 2023-08-16:
@j0ta29 ist es auch möglich den Abfrageintervall zu definieren ?  Mir würde ein einminutiger Intervall reichen..

devganny, 2023-08-16:
> @j0ta29 ist es auch möglich den Abfrageintervall zu definieren ? Mir würde ein einminutiger Intervall reichen..

Da hänge ich mich mal mit dran, mir würde auch ein 60 Sekunden Intervall reichen

j0ta29, 2023-08-16:
@andyyyy2306 und @devganny , das Abfrageintervall könnt Ihr über einen weiteren Eintrag `update_interval`steuern. Beispiel:
```
  - platform: optolink
    name: ${name} Anzahl Brennerstarts
    address: 0x088A
    bytes: 4
    icon: mdi:counter
    state_class: total_increasing
    update_interval: 60s
```
Das geht in ESPHome bei fast alles Komponenten ist aber leider sehr gut versteckt in der Doku.

andyyyy2306, 2023-08-16:
Danke für die Hilfe !

closed 2023-08-16 by andyyyy2306

devganny, 2023-08-16:
> @andyyyy2306 und @devganny , das Abfrageintervall könnt Ihr über einen weiteren Eintrag `update_interval`steuern. Beispiel:
>
> ```
>   - platform: optolink
>     name: ${name} Anzahl Brennerstarts [...]

Vielen Dank für die Arbeit und den Support!

altacoon, 2023-08-25:
Hallo @andyyyy2306 und @devganny ,
ich habe leider immer noch die Probleme, die ihr beide jetzt gelöst habt. In der log-Ausgabe sehe ich auch nur "Read..." und "timeout". Auch bin ich zwischenzeitlich auf einen ESP32 gewechselt, aber ohne dadurch weiter gekommen zu sein.

Nachdem Du von einer kalten Lötstelle berichtet hast, muss ich wohl auch am ehesten (und weiterhin) von einem nicht funktionierenden DIY-Adapter ausgehen. 

Ich hätte mal zwei Fragen bzw. Bitten an Euch:
1.) Mit welcher Version von ESPHome habt ihr kompiliert, mit der "dev"-Version so wie es j0ta29 einmal empfohlen hatte, oder mit der "normalen" ESPHome-Variante?
2.) Könnt Ihr mal Bilder Eures Adaptergehäuses posten. Oft liegt der Fehler ja an einem nicht richtig ausgerichteten Adapter.

j0ta29, 2023-08-25:
@altacoon, die Verwendung der dev-Version war nur temporär notwendig (zwischen 2023.3 und 2023.4), als die ESPHome-Maintainer eine interne Code-Änderung vorgenommen hatten, die dazu geführt haben, dass die `optolink`-Komponente nicht mehr compilierbar war. Kommt selten vor, kommt, aber vor. Und solange neue Komponten auf einem Pull-Request unterwegs sind (d.h. noch nicht offizieller Bestandteil des offiziellen Releases), müssen die Autoren der Komponenten selbst dafür sorgen, dass alles wieder kompatibel ist.

devganny, 2023-08-25:
@altacoon Lässt sich mit allen aktuellen Vesionen compilieren, Du benötigst keine DEV-Version. Ich hatte anfänglich auch Probleme mit der Ausrichtung der Empfänger Diode, hier kam es auf den richtigen Winkel an in dem die Diode auf die Vitodens zeigt. Nach einigem probieren und Druck-Optimierungen habe ich es hinbekommen. Habe die Dioden dann in der richten Position im 3D gedruckten Gehäuse verklebt.

altacoon, 2023-08-26:
Mein dritter DIY-Adapter hat auf Anhieb funktioniert!

Mein Ölverbrauch wird um den Faktor 1000 zu groß angezeigt. Es ist maximal eine Division durch 100 konfigurierbar. Wählt man dies, kommt aber beim Kompilieren ein Fehler. @j0ta29 , ich weiß nicht, ob das ein Hinweis Wert ist, natürlich kann man den Wert auch in ha weiterverarbeiten, das ist mir klar.

j0ta29, 2023-08-26:
@altacoon , danke für den Hinweis. Auf jeden Fall war das den Hinweis wert. Ich hatte in der Tabelle mit den Adressen (https://github.com/openv/openv/wiki/Adressen) die 4 Werte übersehen, die einen Faktor von 1000 haben.
Bin gerade an der Fertigstellung eines neuen Standes, der auf die Konfiguration von Schaltzeiten unterstützt. Dann werde ich auch noch ergänzen, dass der Faktor 1000 unterstützt wird.