openv/openv #610: 1xWW nachladen 0x0247

BerniS84, 2023-10-02:
Hallo,

Wieder mal das Thema 1x Warmwasser nachladen:

die Kommunikation per Optolink Adapter mit meiner Vitocal 300-A (0x0247) läuft seit Jahren stabil. Nun möchte ich die Warmwasserbereitung anstoßen, wenn von der PV überschüssige Energie gelefert wird.

Leider funktionieren die Adressen 0xB020 bzw. 0x65F5 (beide hier in den Issues gefunden) nicht, weder lesend noch schreibend.

Nun die Frage, gibts jemanden der hier dir korrekte Adresse kennt?

Danke schon mal für eure Hilfe!

MrIcemanLE, 2023-10-19:
Hi,

Ich benutze 0xB020 bei meiner Vitocal 300-A. Du musst allerdings den Wert 2 senden:
```
<command name="set1xWW-Bereitung" protocmd="setaddr">
        <addr>B020</addr>
        <len>1</len>
        <unit>RT</unit>
        <description>Bedienung WW Betriebsdaten WW Wertebereich 0 - 2</description>
</command>
```

`set1xWW-Bereitung 2`

BerniS84, 2023-10-27:
Vielen Dank für deine Antwort und die Bestätigung der Adresse.
Dann muss ich mir das nochmal ansehen, ev. stimmt mit meinen BYTE Befehlen noch etwas nicht

MrIcemanLE, 2023-12-07:
Schau mal, ob du den Befehl mit `detail` und `debug on` ausführen kannst. Bei mir sieht das dann so aus. Ich kann mich erinnern, dass es bei mir auch zuerst nicht funktioniert hat und ich irgendwas an den Makros für das P300 anpassen musste. Kann mich aber gerade nicht mehr erinnern. Komisch nur, dass die anderen Befehle funktionieren. Das spricht eher gegen einen Fehler in der vcontrold.xml

```
pi@rpi-heizung:~ $ vclient -c 'detail set1xWW-Bereitung,debug on,set1xWW-Bereitung 2'
detail set1xWW-Bereitung:
set1xWW-Bereitung: SEND 00 02 B0 20 01;SEND BYTES RT;RECV 1 SR
	Bit (BP): 255
	Unit: ReturnStatus (RT)
	  Type: enum
	  Get-Calc: (null)
	                                    Set-Calc: (null)
	 Einheit: (null)
	  Enum Bytes: 00 Text: 0
	  Enum Bytes: 01 Text: 1
	  Enum Bytes: 02 Text: 2
	  Enum Bytes: <default> Text: NOT OK
	Unit: SetReturnStatus (SR)
	  Type: enum
	  Get-Calc: (null)
	                                    Set-Calc: (null)
	 Einheit: (null)
	  Enum Bytes: 00 Text: OK
	  Enum Bytes: 05 Text: SYNC (NOT OK)
	  Enum Bytes: <default> Text: NOT OK
debug on:

set1xWW-Bereitung 2:
DEBUG:Thu Dec  7 09:56:57 2023 : Command: set1xWW-Bereitung 2
DEBUG:Thu Dec  7 09:56:57 2023 : Process 15814 tries to aquire lock
DEBUG:Thu Dec  7 09:56:57 2023 : Process 15814 got lock
DEBUG:Thu Dec  7 09:56:57 2023 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
DEBUG:Thu Dec  7 09:56:57 2023 : Configuring serial interface /dev/ttyUSB0
DEBUG:Thu Dec  7 09:56:57 2023 : >SENT: 04
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 05 (280.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: received 05
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_LINK_STATUS(05) (was FFFF)
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: closed
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 16
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 00
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 00
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 06 (10.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: received 06
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was FE05)
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: opened
DEBUG:Thu Dec  7 09:56:58 2023 : 2 -> 02
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 41
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 06
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 00
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 02
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: B0
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 20
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 01
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: 02
DEBUG:Thu Dec  7 09:56:58 2023 : >SENT: DB
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 06 (20.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: received 06
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: framer_set_actaddr framer_current_addr = 20B0 (was FFFF)
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: Command send
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: no preset result
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 41 (10.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: received 41
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 05 (0.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: received 05
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 01 (10.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 02 (0.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 B0 (0.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 20 (0.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 01 (10.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: len=1 D9 (0.0 ms)
DEBUG:Thu Dec  7 09:56:58 2023 : <RECV: received 01 02 B0 20 01 D9
DEBUG:Thu Dec  7 09:56:58 2023 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was 20B0)
DEBUG:Thu Dec  7 09:56:58 2023 : 00 -> OK
DEBUG:Thu Dec  7 09:56:58 2023 : OK
OK
```

BerniS84, 2023-12-07:
Ich habe noch probiert das 1xnachladen manuell auf der Heizung zu aktivieren, wenn ich die adresse B020 dann lese ist diese immer noch 0. Ist das bei dir auch so?