openv/openv #614: ESPhome with Vitocal 222-G General question and Datapoints

Tobbestoll, 2023-11-07:
Hello,
I am new in this area. I set up a Wemos D1 mini pro with ESPhome using this [explanation](https://github.com/openv/openv/wiki/ESPHome-Optolink).
The Wemos is up and running and I can see it in homeassistant.
At the moment I have only the SFH309 LED soldered. I thought this will be okay to read the values.
I am searching for the other LED meanwhile. 

- is it possible to use only the one LED and read data or must the IR-LED send a command to read it afterwards?
- Which infos do I need to edit the yaml and where can I get them?

I hope someone is willing to help a completely beginner.

Thanks a lot
Tobbes

philverh, 2023-11-07:
If you don't send a command / request the only thing you will be able to read is the regular "Sync" defined by the protocol. No data is sent back without a request. You absolutely need to send a "request" to be able to read some data back.  
And by the way the commands or requests need to be customised to the type of your heating device, that's why the .XML files are so important. They define the communication.

Tobbestoll, 2023-11-08:
Thank you very much for your reply. So I need an IR LED to send commands. I will order two different kinds to test.
What is the easiest request, just to see that my setup works?
Where can I get the information about the commands I have to send? Where can I find these .XML files for my device?

j0ta29, 2023-11-08:
You do not necessarily need the xml files (though they are the best and recent resource to get the information).
Take a look at https://github.com/openv/openv/wiki/Adressen. There you can find for many devices the necessary information (address, bytes and value factors aka div_ratio).
Another good resource is https://github.com/philippoo66/ViessData21/blob/master/DP_Listen_2.zip.
Finally you can install the official Viessmann Software (I think ViessData is the name) and take a look into the installation directory where you find all the xml stuff.

Tobbestoll, 2023-11-08:
Thanks again for the answer.
What is the minimum yaml script I can use to check if the system is running?

Which address may I use first? What do you mean to this first script:

esphome:
  name: vitocal-esp1
  friendly_name: Vitocal-ESP1 

esp8266:
  board: d1_mini_pro

//# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG

external_components:
  - source: github://pr#4453
    components: [ optolink ]

optolink:
  protocol: KW             # P300 oder KW
  logger: enable             # wenn später alles rund läuft, wieder entfernen 
  state: Status              # wenn später alles rund läuft, wieder entfernen
  device_info: 0x00F8 # dient zur Identifizierung der Vitotronic 

//# Enable Home Assistant API
api:
  encryption:
    key: "Zsex1Qxwj+cbWXFcuXzIvUrW2f/CsTAHHXegR0vbR0c="

ota:
  password: "bccf2c796e8b650fc7ca9f28ef55396e"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  //# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Vitocal-Esp1 Fallback Hotspot"
    password: "*******"

captive_portal:

sensor:
  - platform: optolink
    name: ${name} Aussentemp 0x0101
    address: 0x0101
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature

  - platform: optolink
    name: ${name} Aussentemp 0x0800
    address: 0x0800
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature  
`

j0ta29, 2023-11-12:
@Tobbestoll , your config file looks good at first glance.

I would recommend to add the web_server component until your first run succeeds an you need more free memory for more sensors:
```
web_server:
  port: 80
```
With the web_server you should reach your ESP unter http://<ip> and see the log output.

Moreover, if you do not use the debug output on UART1, add baud_rate: 0. This does not affect debug output to the web_server component. 

```
logger:
  hardware_uart: UART1
  baud_rate: 0
  level: DEBUG
```

Tobbestoll, 2023-11-13:
Thanks a lot for all the good hints.
The web server is running, but I have to choose
`web_server:
     local: True`
otherwise I cannot connect to the wemos.

Now I have made a wooden plate for the LEDs. I am using a SFH309-5/6 (without FA) and a SFH4356.
Unfortunately both with 3mm diameter. It is very difficult to place them in the right way.
The SFH4356 needs to be a little bit away from the LED inside the heater. I am powering the IR with 5v instead of 3.3V.
Hope it will run a little bit.

With the help of the web server I was able to get some answers from the heater.
Now I am searching the addresses.
I do not understand how I can extract them from the software or elseware.
Can you help me, for my Vitocal 222G?

Thanks Tobbes

j0ta29, 2023-11-13:
With your config...

```
optolink:
  ...
  device_info: 0x00F8 # dient zur Identifizierung der Vitotronic
```
...you should see a text sensor with the name `0x00F8` (I recommend a more meaningful name ;-)). Something like

```
Device ID: 20C2|Hardware Revision: 00|Software Index: 06
```

The important part is the value after 'Device ID', in my case `20C2`. This is the identification of the controlling unit. In your case it's for sure a other value.
In the installation directory of the Viessmann Software you find a lot of XML files.
In my case (I don't know if the software still is the same) there is a file named ".../VitosoftConfig/configbackup/ecnDataPointType.xml" where I search for `20C2` and find a text block with many addresses:

```
    <Identification>20C2</Identification>
    <Options>undefined</Options>
    <EventTypeList>ecnStatusEventType;AE_Eingang0_10~0x0A86;AE_Eingang1~0x0A80;AE_Eingang2~0x0A81;Ausgang_HKP_A1~0x7663;Ausgang_HKP_M2~0x7665;BauartWarmwasser~0x8851;Bedien_WW_Solltemperatur~0x6300;BedienparameterA1M1FunktionReset~0xA051;BedienparameterM2FunktionReset~0xA051;BedienPartybetriebA1M1~0x2303;BedienPartybetriebM2~0x3303;BedienRTSolltemperaturA1M1~0x2306;BedienRTSolltemperaturM2~0x3306;BedienRTSolltemperaturReduziertA1M1~0x2307;BedienRTSolltemperaturReduziertM2~0x3307;BedienSparbetriebA1M1~0x2302;BedienSparbetriebM2~0x3302;Bedienteil_SW_Index~0x7330;
```
The parts after the `~`  (e.g. 0x0A86) are the addresses of the so called datapoints.
There are more XML files in the installation directory (in my case the file ecnEventType.xml), where you find more information about every datapoint.

Tobbestoll, 2023-11-14:
Thank you very much for the support.
Unfortunately my Device ID is not shown in the xml files of the software.
My device ID is **204B**. At the moment it is more trail and error to find some addresses.
Is there an other source? 

I would like to implement control elements like "room temperature" , "water temperature" , " one time water heating" and "Sparbetrieb ;-) ".
How do add these in my yaml?

j0ta29, 2023-11-16:
@Tobbestoll , unfortunately there isn't any information about your device in the resources that I know.
I found a question about the 204B device on the Viessmann community site (https://www.viessmann-community.com/t5/Waermepumpe-Hybridsysteme/aktuelle-KNX-Datenpunkte-204B-Vitocal-xxx-G-Vitotronic-200-Typ/td-p/364289) but unfortunately the guy has the same problem like you and did not get an answer yet.

Tobbestoll, 2023-11-17:
@j0ta29 thanks for your answer. I found a lot of info here [link](https://github.com/openv/openv/issues/415).
I could add to home assistant:
temp outside 0x0101
boiler temp soll 0x6000
boiler temp ist 0x010D
room temp soll 0x2000

But I am not able to control the room temperature or boiler temp over homeassistant. My entries are ignored.
My wishes are for home assistant:
A switch to start Warmwater heating once
Start the "Sparbetrieb"
See if the pump for circulation is running
See the COP values
Change room and boiler temp

Tobbestoll, 2023-11-20:
Hello all,

it would be cool to control my Vitocal if my photovoltaik produce enough energy.
So I try to start the one time water heating over openv. I edited the "temp2" value some degrees over the normal water temperature.

Here is my yaml code which will not start the heating.
Is there something wrong with the code or could it be the wrong address?

`switch:
  - platform: optolink
    name: WW_Bereiten_einmal
    address: 0xB0D0
    icon: mdi:mode-heat-outline`

Is switch the correct type?

Thanks for your help.

My second problem is the controlling of the room temperature.
I found the address to see the "soll" temperature but I cannot change this value.
Here my code for this:

`number:
  - platform: optolink
    name: ${name} Sollwert Raumtemperatur
    unit_of_measurement: °C
    address: 0x2000
    bytes: 2
    div_ratio: 10
    min_value: 3
    max_value: 37
    step: 1
    mode: box
    icon: "mdi:home-thermometer"
    device_class: temperature`

Many thanks for a hint!

j0ta29, 2023-11-21:
@Tobbestoll , If the addresses are correct for your device, this configuration looks good.

`switch` and `number` are the active (aka read-write) counterparts to the passive (ake read-only) components `boolean_sensor` and `sensor`.
Did you try to configure the passive components (in addition to `switch` and `number`) to check if the values where accepted by the Viessmann device?