openv/openv #615: Probleme mit Installation von Vcontrold

Hoizi007, 2023-11-07:
Hallo Leute,

Habe leider das Problem dass ich bei der Installation nicht weiter komme bin nach der Anleitung hier gegangen habe allerdings allerdings andere pfade zb. bei mir liegen die xml datein in openv/vcontrold/xml/vito.xml als beispiel was ist hier schon falsch gelaufen das ich hier ander pfade habe? Wie fahre ich mit punkt 6 vor habe das Aktuelle debian bookworm oben wo erstelle ich das startup script bzw hüpfe ich gleich auf punkt 6.1 weiter? wenn ja lässt sich das ganze auch nicht mit sudo /etc/init.d/vcontrol start starten.

irgendwie hab ich komplett den überblick verloren hat evtl. jemand noch eine kleine Anleitung für dummys :) oder tipps?

lg.

philverh, 2023-11-07:
The XML files need to be configured for your heating device and setup choices such as the name of the IR interface. The 2 configured .xml files need to be in /etc/vcontrold.

More instructions are available on the Wiki space in this project. For Raspberry (debian based) look at https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi

Hoizi007, 2023-11-07:
yes i am hanging at point 6 how do i start vcontrold i hope so far its right atm at before point 6

philverh, 2023-11-07:
Skip step 6 and do what is listed under 6.1. 
6 (.0) does not work anymore in the more recent releases of raspbian/Debian. 

I called my file "vcontrold.service" (with an additional "d") and adapted the other commands accordingly.

I just re-did a full install last weekend with, and it works. I did try to get it working with the latest release (Bookworm/12) but gave up when I noticed VNC does not yet work. So I fell back on the release based on "Bullseye/11" and all works.

Hoizi007, 2023-11-08:
so i have installed all now i just wantet to start the test 

now i get this

pi@raspberrypi:~ $ telnet localhost 3002
Trying ::1...
Trying 127.0.0.1...
telnet: Unable to connect to remote host: Connection refused

any ideas?

philverh, 2023-11-08:
Have you rebooted your device? 
Try issuing a "sudo systemctl status vcontrold.service" to make sure the service is actually running. If you rebooted and the status is not Active/Running, this means there is something wrong with the set-up of the autostart of the service. 
You could also issue "sudo systemctl start vcontrold.service" to see what is happening.
When the service is running, running "telnet localhost 3002" should answer with a prompt "vctrld>" where you could for example type "version" to get the running version string, or "commands" to list all recognised commands from your .xml files. 
Type "quit" when you are done ;-)

Hoizi007, 2023-11-08:
oh there is a problem with the setup 

pi@raspberrypi:~ $ sudo systemctl status vcontrold.service
● vcontrold.service - Viessmann heating systems interface
     Loaded: loaded (/etc/systemd/system/vcontrold.service; enabled; vendor pre>
     Active: failed (Result: exit-code) since Wed 2023-11-08 12:15:13 CET; 2h 3>
       Docs: https://github.com/openv/vcontrold
    Process: 509 ExecStart=/home/pi/openv/startup.sh (code=exited, status=203/E>
   Main PID: 509 (code=exited, status=203/EXEC)
        CPU: 5ms

Nov 08 12:15:13 raspberrypi systemd[1]: vcontrold.service: Scheduled restart jo>
Nov 08 12:15:13 raspberrypi systemd[1]: Stopped Viessmann heating systems inter>
Nov 08 12:15:13 raspberrypi systemd[1]: vcontrold.service: Start request repeat>
Nov 08 12:15:13 raspberrypi systemd[1]: vcontrold.service: Failed with result '>
Nov 08 12:15:13 raspberrypi systemd[1]: Failed to start Viessmann heating syste>
lines 1-13/13 (END)
● vcontrold.service - Viessmann heating systems interface
     Loaded: loaded (/etc/systemd/system/vcontrold.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-11-08 12:15:13 CET; 2h 37min ago
       Docs: https://github.com/openv/vcontrold
    Process: 509 ExecStart=/home/pi/openv/startup.sh (code=exited, status=203/EXEC)
   Main PID: 509 (code=exited, status=203/EXEC)
        CPU: 5ms

Nov 08 12:15:13 raspberrypi systemd[1]: vcontrold.service: Scheduled restart job, restart counter is at 5.
Nov 08 12:15:13 raspberrypi systemd[1]: Stopped Viessmann heating systems interface.
Nov 08 12:15:13 raspberrypi systemd[1]: vcontrold.service: Start request repeated too quickly.
Nov 08 12:15:13 raspberrypi systemd[1]: vcontrold.service: Failed with result 'exit-code'.
Nov 08 12:15:13 raspberrypi systemd[1]: Failed to start Viessmann heating systems interface.

philverh, 2023-11-08:
This is what I have in /etc/systemd/system/vcontrold.service:
---
[Unit]
Description=vcontrold for Viessmann
Documentation=https://github.com/openv/vcontrold
After=syslog.target systemd-udev-settle.service

[Service]
Type=forking
ExecStart=+/usr/local/sbin/vcontrold -x /etc/vcontrold/vcontrold.xml --inet4 --syslog
StandardOutput=inherit
StandardError=inherit
ExecReload=/bin/kill -HUP $MAINPID
Restart=on-failure
RestartSec=10
StartLimitBurst=5
#StartLimitIntervalSec=30

[Install]
WantedBy=multi-user.target
---

I have no /home/pi/openv/startup.sh 

Hope this helps.

Hoizi007, 2023-11-08:
OMG its works with a little bit adjustment :D im so happy