openv/openv #617: Vitocal 200 S and ESPHome connection

budisek, 2023-11-15:
I am trying to connect my Vitocal 200 S via ESPhome to Home assistant.

IR LED is blinking on phone camera but I cannot check that I have correct connection.
Is this right code for check some incomming temperature values?

```
esphome:
  name: "nodemcu1"
  friendly_name: "nodemcu1-vitocal"

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  hardware_uart: UART0
  baud_rate: 0

...

external_components:
  source: github://pr#4453
  components: [ optolink ]

optolink:
  protocol: P300
  device_info: Device Info
  state: Component state

sensor:
  - platform: optolink
    name: ${name} Heater Temperature
    address: 0x0810
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature

  - platform: optolink
    name: ${name} Outside Temperature
    address: 0x0101
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature
```

j0ta29, 2023-11-16:
@budisek, your config file looks good at first glance. I would recommend to add the web_server component until your first run succeeds and you need more free memory for more sensors:
```
web_server:
  port: 80
```
With the web_server you should reach your ESP unter http:// and see the log output.

budisek, 2023-11-16:
Yes I tried add web_server before.

But I dont know how to check that ESPHome talk with the heater correctly. I watch the log but there are no incomming values.

j0ta29, 2023-11-20:
@budisek , if you do not see any log entries or usefull values in der web_server interface, I suggest to check your DIY adapter. 
Take a look at https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702. User derwassi describes how to check the function of your adapter with a simple ESPHome configuration where the adapter LED sends signals directly to its own photo transistor.

budisek, 2023-11-24:
I have found it. It was bad polarity of read IR LED.

May I ask where can I find all available addresses for Vitocal 200 S? I cannot find it on the list.

j0ta29, 2023-11-24:
@budisek , glad to hear that you found the failure.

The only resources I know are
- https://github.com/openv/openv/wiki/Adressen
- https://github.com/philippoo66/ViessData21

If you cannot find any information there, get the code of your device (textsensor 'Device Info', first part) and search with 'datapoint' or 'viessmann'.

Good luck!

budisek, 2023-11-24:
`Device ID: 204D|Hardware Revision: 01|Software Index: 41`

Is this correct? I can't find my device in any of both ZIP archives.

j0ta29, 2023-11-24:
I forgot to say "... in the internet" ;-)
Google "204D viessmann datapoints"
This seams to be a promising result: https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwi53-PAnN2CAxUmhf0HHVUhBvUQFnoECBUQAQ&url=https%3A%2F%2Fwww.viessmann-community.com%2Fviessmann%2Fattachments%2Fviessmann%2Fcustomers-connectivity%2F45542%2F1%2Fvito.xml.pdf&usg=AOvVaw1lf1lY4lk9NfaN_LJ7S_6n&opi=89978449