openv/openv #624: Segmentation fault (Vitotronic 300 (KW3))

b3nsen, 2023-12-29:
[myConfig.zip](https://github.com/openv/openv/files/13795811/myConfig.zip)
[log_vcontrold.txt](https://github.com/openv/openv/files/13795829/log_vcontrold.txt)

Hallo Zusammen,

ich bekomme immer einen 'Segmentation fault' wenn ich '/usr/sbin/vcontrold' ausfuehre. 

Ich habe einen Raspberry Pi Zero W und einen "fertigen" USB-Adapter von Dirk (https://github.com/openv/openv/wiki/Bauanleitung-USB)

vcontrold wurde auf dem Raspberry kompiliert, habs auch schon mit einem fertigen Paket versucht was auch nicht funktionierte.

Die Dateien meiner Config fuer den Vitotronic 300 (KW3) habe ich angehaengt. 

Vielen Dank fuer eure Hilfe

daDomas, 2024-01-01:
Hallo,

vorab, ich bin selber Leihe in dem Bereich und arbeite mich ebenfalls mit einem aktuellen PI sowie einem vorkonfektioniertem Kabel ein.

Segmentation fault hatte ich auch.
Versuche mal folgende Option:
`vcontrold -n`

Dann solltest du bereits etwas mehr sehen. Ich hatte Probleme beim Port. Der default Port 3002 war bereits belegt. Ferner hatte ich Probleme bei Zugriffen auf das Log-File. Hier kannst du mit der Option -l einen alternativen Pfad, z.B. in deinem Homeverzeichnis angeben.

Viel Erfolg

sina66, 2026-08-09:
Hallo ihr 2 

Wie sieht denn eure Adressliste für die Vitotronic aus?
Selbst habe ich die Vitotronic 300 Typ KW3. 

Ich schaffe es leider noch nicht mit dem Auslesen. Ich tippe auf die falschen Adressen, denn die Technik funktioniert (senden und empfangen)

Viele Grüße!

PS
Wenn ich mit einem reinen Receiver auslese was von der Heizung gesendet wird, dann kommt immer wieder das:
[[remote.pronto:239]: 0000 006D 0004 0000 0006 0003 000F 0001 000F 0001 0037 0180

Ich kann damit leider nichts anfangen.