openv/openv #625: vito.xml for Vitodens 100 (GWG_BT)

Bascht74, 2024-01-05:
Hallo, 

ich beschäftige mich erst seit kurzem mit vcontrold, da mein Freund eine Vitodens 100 hat, die nicht über VitoConnect läuft, da sie nur das alte GWG-Protokoll unterstützt. 

Ich habe herausgefunden, dass die Vitodens 100 den Gerätetyp 2054 hat (GWG_BT).
Die vito.xml, die bei vcontrold mit dabei ist, hat allerdings nur für die 2053 eine Konfiguration dabei. 
Für die 2054 scheinen unterschiedliche Adressen notwendig zu sein. 

Hat jemand hier eine funktionierende vito.xml für die aktuelle Version von vcontrold für den Gerätetyp 2054?
Aktuell bin ich noch ein wenig lost ...

Falls mir jemand helfen kann, wäre das toll. 
Vielleicht kann ich dann bei meinem Freund herausbekommen, was bei seiner Heizung schief läuft...

Bascht74, 2024-01-08:
Vielleicht habe ich das Ticket an der falschen Stelle erstellt. 
Daher nochmal hier direkt unter vcontrold in der Hoffnung, dass noch jemand Informationen / Daten hat:
https://github.com/openv/vcontrold/issues/138

Soll ich das Ticket dann hier schließen?

philverh, 2024-01-08:
I was able to find device 2054 in https://github.com/philippoo66/ViessData21/tree/master
Not in the direct XML format, but it still contained a series of addresses which may help you.

Bavarese, 2024-01-12:
Betreibe selbst eine Anlage mit GWG-Protokoll. Hier ein Weg, um zu passenden Datenpunkten zu kommen (aus dem Netz, ohne Gewähr oder weiteren Support) -

Laden Sie sich bei Vissmann die 90 Tage laufende Demo der PC-Software (für Windows) herunter. Installation am Laptop. Sie benötigen in der Regel noch ein Optolink-Kabel (das man nicht vom Hersteller kaufen) muß, sowie tatsächlichen Zugang zur Heizung. 

Das Optolink-Kabel wird in die V-förmige Aussparung gesetzt (zu finden in der Nähe des Bedienpanels, ggf. unter einer Klappe). 

Achtung: nach dem Start läuft die Software 90 Tage als Vollversion, bis dahin sollten Sie alle Abfragen erledigt haben und auch via Optolink Kontakt zur Heizung aufgenommen haben (Abfrage).

Machen Sie sich Notizen und sehen Sie sich dann die erzeugten XML-Dateien auf Ihrem PC genau an. Interessant sind Datenpunkte und deren Beschreibung.

Beispiel : "Einmalige WW-Bereitung (0xB020: 0:Manuell, 1:WW-Komfort)"
(willkürliches Beispiel, meine Heizung unterstützt B020 nicht)

Link:
https://www.viessmann-community.com/t5/Waermepumpe-Hybridsysteme/Inbetriebnahme-Software/td-p/364334

Es gibt auch Python-Programme zur übersichtlichen Aufbereitung der Quelldaten (hier kein Link).

Sie können natürlich auch den freundlichen Heizungsbauer fragen ;-)