openv/openv #627: Zweck "Optimierung" konkret?

bibcoder, 2024-01-09:
Hallo Leute,

ich habe auch eine Viessmann Ölheizung Vitoladens 300C + Vitronic und würde diese gerne aus Spaß mal anzapfen, schon alleine um die Temperatur zu loggen. Beim Lesen der Wikiseiten stellte sich mir aber erstmal noch eine andere Frage, da steht nämlich als Motivation für dieses Projekt neben "Überwachung, Protokollierung, Fernsteuerung, Analyse" auch "Optimierung".

Deswegen würde mich mal interessieren, ob es auch Erfahrungsberichte darüber inwieweit die zusätzlich gewonnen Daten bereits auch dazu genutzt werden konnten den Heizverbrauch zu senken. Dazu fand ich aber im Wiki nichts. Gibts dazu vielleicht so ne Art allgemeines Vorgehen... à la "die und die Werte überwacht... und dann die Heizkurve angepasst" etc?

Bitte nicht falsch verstehen, ich find das schon auch toll, wenn ich die Heizung auslesen kann und nicht von Viessmann so abhängig bin, aber es wäre natürlich auch hilfreich, um meiner Frau zu erklären, dieses Projekt anzufangen, wenn ich sagen könnte, dass das dann 10% Heizkosten spart oder so ;-)

Beste Grüße
Bibcoder

jaegerschnitzel, 2024-01-10:
Ich glaube das Hauptproblem ist, dass man das nie genau beziffern kann, da alle Parameter variabel sind. Der Verbrauch hängt ja von vielen Dingen ab, die sich aber ständig ändern. Beispielsweise, Außentemperatur, Warmwasser, Sonnenscheindauer, Lüftung, Kochen, etc, etc.
Aber grundsätzlich können die Daten natürlich helfen, um die Parameter der Heizung zu optimieren und z.B. die Taktung zu optimieren.

bibcoder, 2024-01-10:
Ah verstehe, dass sich konkrete Einpsparpotenziale nicht vorher abschätzen lassen. Ich hab grad nach "Heizung" und "Taktung" gegoogelt, so ein Laie bin ich :)
Hab da auch gleich nen schönen Artikel gefunden dazu, den ich mir heut Abend gleichmal durchlesen.

Gibts denn irgendein Buch, Link oder anderes Forum, wo man dann vielleicht so als IT-Profi aber Heizungs-Laie ne grobe Idee bekommt, welche Werte man mittels OpenV sich holen sollte, über einen längeren Zeit beobachten sollte, um dann die... z.B. Taktung zu optimieren?

Denk, das würde einigen Helfen so nen Ziel vor den Augen zu haben (z.B. Taktung optimieren) und dann mittels diesen schönen OpenV umsetzen zu können

j0ta29, 2024-01-10:
Bei mir hat schon geholfen, dass ich aus den gemessenen Werten die Heizung und ihre aktuelle Einstellung besser verstehen konnte. 
Beispiel: In unserem Haushalt die Brauchwassertemperatur bis 22 Uhr "oben" zu halten macht keinen Sinn, weil die ganze Familie morgens unter der Dusche steht. Also wird bei uns jetzt nur noch morgens das Wasser aufgeheizt. Die Temperatur fällt im Laufe des Tages zwar langsam ab, aber diese "Resttemperatur" ist in der Regel ausreichend. Falls nicht, wird per Handy der Partybetrieb angeschaltet.
Diese Optimierung hat schon mal spürbar Gas gespart. Den Gasverbrauch nebenbei zu messen, macht natürlich auch Sinn um die Auswirkung einer Änderung einschätzen zu können.

bibcoder, 2024-01-13:
Danke @j0ta29 für deine Antwort. Das mit den Einstellungen kann ich gut nachvollziehen, ich versteh das Ganze auch noch nicht so wirklich und deine Antwort hat mich dazu motiviert, das doch mal anzugehen, Danke dir und @jaegerschnitzel dafür! :-) 👍 👍 👍

j0ta29, 2024-01-13:
Und falls du nichts zum Optimieren findest, dann hast du zumindest das gute Gefühl gefunden, dass alles schon optimal ist.
Und vor allem: Du hattest ja geschrieben "gerne aus Spaß mal anzapfen". Spaß ist doch auch ein Argument, das deine Frau
nicht so einfach wegdiskutieren kann, oder ;-) ?

bibcoder, 2024-01-13:
> Und falls du nichts zum Optimieren findest, dann hast du zumindest das gute Gefühl gefunden, dass alles schon optimal ist. Und vor allem: Du hattest ja geschrieben "gerne aus Spaß mal anzapfen". Spaß ist doch auch ein Argument, das deine Frau nicht so einfach wegdiskutieren kann, oder ;-) ?

Hehe... ja Spaß macht mir das bestimmt... nur meine Frau meint, ich sollte lieber Ihr Auto reparieren... ;-)