openv/openv #628: Serieller Optolink-Adapter -- beste Bauvariante für längere Strecke?

bibcoder, 2024-01-13:
Schönen Samstag zusammen!

ich bin am Überlegen, welche Optolink-Adapter-Bauanleitung aus dem Wiki denn für mich am besten ist. Es sollte relativ einfach sein, weil ich eher aus der Software-Ecke komme und kein Elektrotechnik-Profi bin. 

Außerdem sind da noch die Gegebenheiten unseres Heizungskellers: Eine Steckdose gibts da zwar schon, aber da hängt der Trockner dran. Die Heizung selbst ist festangeschlossen an so ein quadratisches Kästchen, welches nur Elektriker installieren dürfen. 
Außerdem steht ein Zimmer weiter mein Heimserver, zirka 15-20m weiter. Allerdings ist da eine Wand dazwischen, könnte man aber noch eine Leitung irgendwie unterhalb vom Heizungsrohr reinquetschen.

**Kurzfassung:**
Wegen dem Server im Nachbarzimmer und der schlechten Stromanbindung im Heizungsraum, scheint es mir fast sinnvoller so einen 3,3V-Optolink-Adapter, welcher mit altem Telefonkabel und ohne Netzteil funktioniert, umzusetzen als so einen "modernen" USB-Adapter. 

**Frage:**
Was sagen die erfahrenen Hasen hier dazu? -- ist in diesen Umständen die 3,3-TTL-optolink-Variante die beste oder habe ich da irgendwas übersehen?

Schönes Wochenende
Bibcoder

JeemainGitHub, 2024-01-18:
Hi Bibcoder,

ich habe mir den OptoLink USB-Duo/Sniffer zu Hause nachgebaut, um die Signale zu loggen und gleichzeitig mit dem Viessmann-Server verbunden zu sein. Die Heizung steht im Keller und mein Sniffer ist ca. 15 m entfernt im EG untergebracht. Signaltechnisch habe ich keine Probleme.

Aufbau:
Sensorplatine und Elektronikplatine sind, wie in der Anleitung getrennt, d.h. auf der Sensorplatine befinden sich lediglich der Fototransistor und die IR-Led. Als Leitung verwende ich S/FTP Ethernetkabel. Auf einem Adernpaar liegt TX und 3V3, auf einem zweiten Paar RX und 3V3. An die Sensorplatine selber ist ein 2m langes Telefonkabel (flach) eingegossen, auf der anderen Seite gefindet sich ein Stecker zur Verbindung mit der Ethernetleitung. Somit kann ich für Wartungszwecke alles schnell demontieren. Im EG ist das Ethernetkabel an ein Patchfeld angeschlossen. Per Patchkabel kann man den OptoLinkUSB-Duo/Sniffer dann direkt anschließen.

Wenn du die Funktionalität des Sniffers nicht benötigst kannst du den einen Zweig ja weglassen.

vich-667, 2024-01-18:
Ich nutze einer Erweiterung des OptoLink USB-Duo/Sniffer https://github.com/vich-667/OptolinkDuoBlock so dass ich die Vitoconnect blockieren und direkte mit der Heizung kommuniziren kann. Bei mir ist die Komplette Schaltung incl. der USB Serial Patine für neben der Heizung. Die andere USB Seriell Platine steck an meinem Raspi und da sind ca. 5m Kabel dazwischen (auf dem Schaltplan J1. Das funktioniert seit fast einem Jahr sehr stabiel! Zuerst hatte ich auch die USB Serial Platine für den Raspi an der Heizung, aber das 5m USB Kabel war sehr problematisch.