openv/openv #629: Verkauf von nicht mehr benötigten Teilen zur Kommunikation über Optolink mit / Selling of parts that are no longer needed for communication via Optolink with a Vitodens 200 WB2einer Vitodens 200 WB2

GitHubTW, 2024-01-13:
Hallo und high everybody,

möchte jemand seine gebrauchten Teile verkaufen die für die Kommunikation über die Optolink-Schnittstelle mit einer Vitodens 200 WB2 benötigt werden?

Would anyone like to sell their used parts that are needed for communication via the Optolink interface with a Vitodens 200 WB2?

Vielen Dank und kind regards
Thomas

daDomas, 2024-02-04:
Hallo Thomas,

was genau suchst du? Ich habe noch einige USB Kabel, original, hier liegen.

Viele Grüße, Thomas

bibcoder, 2024-02-08:
Hallo "daDomas"

> was genau suchst du? Ich habe noch einige USB Kabel, original, hier liegen.
> Viele Grüße, Thomas

was genau für Teile benötigst du nicht mehr und wieviel willstn noch ungefähr dafür haben?
Schöne Grüße

GitHubTW, 2024-02-11:
@daDomas 

Hallo Thomas
Sorry dass ich mich jetzt erst melde. Ich hoffe bibcoder hatn mir jetzt nicht alles weggeschnappt!
😱

Was ich wirklich genau brauche weiß ich noch gar nicht...

Ich habe eine Viessmann WB2 mit der Optolink-Schnittstelle (ich denke so heisst das Ding auf Bild unten) und möchte gerne ein  über einen RPI4B ein paar Daten (Temperatur, Brenner AN/AUS...) aus der Heizung auslesen.
Entweder direkt über eine USB-Verbindung zur o.g. Optolink.Schnittstelle oder wenn möglich über WLAN (Optolink > WLAN > RPI).

Bin für alle Infos dankbar!

Viele Grüß
Thomas

[image]

bibcoder, 2024-02-12:
> @daDomas
> 
> Hallo Thomas Sorry dass ich mich jetzt erst melde. Ich hoffe bibcoder hatn mir jetzt nicht alles weggeschnappt! 😱

Doch! Hat er! Und alles schon vermampft! 🥳

... glaub DoThomas hat ja eh geschrieben, er hat mehrere... also chill mal. Wird sich irgendwann schon noch melden.

Schönen Fasching beisammen :)