openv/openv #634: Gehäuse ESP8266 / OpenV

JWSK81, 2024-02-18:
Moin moin meine Herren,

vielen Dank für das gut dokumentierte Projekt.

Ich habe für die Vitotronix 200-H ein Gehäuse entworfen
um den ESP hintder der Abdeckklappe betreiben zu können.

Vielleicht braucht es ja jemand:
https://www.thingiverse.com/thing:6491575

JWSK81, 2024-02-18:
So, 

ich habe jetzt noch ein V-mount Gehäuse gezeichnet, gedruckt und verdrahtet - funktioniert auch :)

adx-git, 2024-04-04:
Cooles design, das werde ich versuchen nachzubauen, definitiv besser als meine fliegende Verdrahtung.
Welche Dioden verwendest du? Die Empfohlenen sind EOL und ich trau mich nicht auf gut Glück eine andere zu verwenden.
Hast du vielleicht sogar einen Link zu Ali?
Danke!

JWSK81, 2024-04-04:
Moin, die verfügbaren LEDs stehen direkt bei der passenden Bauanleitung:
https://github.com/openv/openv/wiki/ESPHome-Optolink

adx-git, 2024-04-04:
Ich drucke mal beide und schau dann mit was ich besser zurecht komme.
Das Erste finde ich schön aufgeräumt, das zweite ist super kompakt.

Wenn ich Fragen zu den Designs habe, werde ich in Thingiverse fragen :-)