openv/openv #635: ESPHome - Failed config - [device_info] / [state] is an invalid option

ChrizZz90, 2024-02-27:
I don't get why HA can't compile the yaml file and install it. I get this mistake:

```
INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/heizung-d1mini.yaml...
Failed config

optolink: [source /config/esphome/heizung-d1mini.yaml:18]
  protocol: KW
  
  [device_info] is an invalid option for [optolink]. Please check the indentation.
  device_info: Viessmann
  logger: enable
  
  [state] is an invalid option for [optolink]. Please check the indentation.
  state: Status
```

This is the yaml file:
```
esphome:
  name: heizung-d1mini
  friendly_name: heizung-d1mini

esp8266:
  board: d1_mini

logger:
  hardware_uart: UART1
  level: DEBUG
  baud_rate: 0  

external_components:
  source: github://pr#4453
  components: [ optolink ]

optolink:
  protocol: KW
  device_info: Viessmann
  logger: enable
  state: Status

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: 
  password: 

text_sensor:
  - platform: optolink
    mode: DEVICE_INFO
    name: Device Info
    update_interval: 15s
  - platform: optolink
    mode: STATE_INFO
    name: Component state
```

I still believe I followed the instructions, but probably there is something I didn't see :)

j0ta29, 2024-02-28:
Please take a look at https://github.com/esphome/esphome/pull/4453#issuecomment-1951339702 and the updated documentation at https://github.com/esphome/esphome/pull/4453#issuecomment-1951339702.
Unfortunately there were some breaking changes in the newest release.

ChrizZz90, 2024-02-28:
ah alright, I mixed both versions. But still with the adapted jaml I don't get a real Device name:

```
esphome:
  name: heizung-d1mini
  friendly_name: heizung-d1mini

esp8266:
  board: d1_mini

logger:
  hardware_uart: UART1
  level: DEBUG
  baud_rate: 0  

external_components:
  source: github://pr#4453
  components: [ optolink ]

optolink:
  protocol: P300
  logger: enable

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: 
  password: 

text_sensor:
  - platform: optolink
    mode: DEVICE_INFO
    name: Device Info
    update_interval: 15s
  - platform: optolink
    mode: STATE_INFO
    name: Component state
```

the logs are:

```
INFO ESPHome 2024.2.1
INFO Reading configuration /config/esphome/heizung-d1mini.yaml...
INFO Starting log output from 192.168.2.190 using esphome API
INFO Successfully connected to heizung-d1mini @ 192.168.2.190 in 0.015s
INFO Successful handshake with heizung-d1mini @ 192.168.2.190 in 3.137s
[12:50:06][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 28 2024, 12:36:23
[12:50:06][C][wifi:577]: WiFi:
[12:50:06][C][wifi:409]:   Local MAC: 
[12:50:06][C][wifi:414]:   SSID: [redacted]
[12:50:06][C][wifi:415]:   IP Address: 
[12:50:06][C][wifi:416]:   BSSID: [redacted]
[12:50:06][C][wifi:418]:   Hostname: 'heizung-d1mini'
[12:50:06][C][wifi:420]:   Signal strength: -87 dB ▂▄▆█
[12:50:06][C][wifi:424]:   Channel: 11
[12:50:06][C][wifi:425]:   Subnet: 
[12:50:06][C][wifi:426]:   Gateway: 
[12:50:06][C][wifi:427]:   DNS1: 
[12:50:06][C][wifi:428]:   DNS2: 
[12:50:06][C][logger:447]: Logger:
[12:50:06][C][logger:448]:   Level: DEBUG
[12:50:06][C][logger:449]:   Log Baud Rate: 0
[12:50:06][C][logger:451]:   Hardware UART: UART1
[12:50:07][C][mdns:115]: mDNS:
[12:50:07][C][mdns:116]:   Hostname: heizung-d1mini
[12:50:07][C][ota:096]: Over-The-Air Updates:
[12:50:07][C][ota:097]:   Address: heizung-d1mini.local:8266
[12:50:07][C][ota:100]:   Using Password.
[12:50:07][C][ota:103]:   OTA version: 2.
[12:50:07][C][api:139]: API Server:
[12:50:07][C][api:140]:   Address: heizung-d1mini.local:6053
[12:50:07][C][api:142]:   Using noise encryption: YES
[12:50:11][I][optolink:058]: requesting value of datapoint Device Info

[12:50:11][E][optolink:060]: read request rejected due to queue overload - queue size: 2
[12:50:11][D][optolink:062]: queued datapoint: Device Info
[12:50:15][D][text_sensor:064]: 'Component state': Sending state 'OK'
[12:50:25][D][text_sensor:064]: 'Component state': Sending state 'OK'
[12:50:26][I][optolink:058]: requesting value of datapoint Device Info

[12:50:26][E][optolink:060]: read request rejected due to queue overload - queue size: 2
[12:50:26][D][optolink:062]: queued datapoint: Device Info
[12:50:35][D][text_sensor:064]: 'Component state': Sending state 'OK'
```

j0ta29, 2024-03-01:
The log output `read request rejected due to queue overload` is a signal that your DIY adapter is not working correctly.
ESPHome/optolink is still waiting for a response from the serial interface in response to the first read request.
If the response is outstanding no more read request are initiated.
Here is an example log for a working communication:
```
[14:45:25][I][optolink:058]: requesting value of datapoint Gastherme Gerätekennung
[14:45:25][I][optolink.datapoint_component:051]: recieved data for datapoint Gastherme Gerätekennung: 100712992
[14:45:25][D][text_sensor:064]: 'Gastherme Gerätekennung': Sending state 'Device ID: 20C2|Hardware Revision: 00|Software Index: 06'
```

ChrizZz90, 2024-03-01:
aaah, perfect - thanks a lot - now it's working

closed 2024-03-01 by ChrizZz90