openv/openv #639: Vitodens 222-F and ESPHome Viessmann Optolink

RevLaw, 2024-04-05:
Hello openv Guys,

first, thanks for all the work you are doing here, it's great.

I have a problem with my DIY optolink ESP32. It is not getting any info out of the optolink interface.
My Board is a: ESP 32 D1 Mini with an ESP-WROOM32 chip
I Checked the IR LED. It is flashing and seems to send signals. The polarisation seems right and all connections on the board are working.
I would like to check if my PT LED is working, but I don't know how.

Also, could it be that my optolink interface of the "Vitodens 222-F" is not supported because I see no ID or Protocol in this [table](https://github.com/openv/openv/wiki/Ger%C3%A4te)

This is my configuration for ESPHome:

```substitutions:
  name: esphome-web-9833f0
  friendly_name: OptolinkReader

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

wifi:
  # Set up a wifi access point
  ap: {}

web_server:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

external_components:
  - source: github://pr#4453
    components: [ optolink ]

optolink:
  protocol: P300           # or do i need to use KV?
  logger: true               # also is this line right because in the esphome preview it says boolean but in the openv wiki 'enabled'?
  rx_pin: 1
  tx_pin: 3

text_sensor:
  - platform: optolink
    type: DEVICE_INFO
    name: Device Info
    update_interval: 10s
  - platform: optolink
    type: STATE_INFO
    name: Component state
    update_interval: 20s```

And this is the logging output:

Time | level | Tag | Message
-- | -- | -- | --
01:41:33 | [D] | [text_sensor:064] | 'Component state': Sending state 'OK'
01:41:42 | [W] | [optolink.datapoint_component:138] | 2. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:41:52 | [W] | [optolink.datapoint_component:138] | 3. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:41:53 | [D] | [text_sensor:064] | 'Component state': Sending state 'OK'
01:42:02 | [W] | [optolink.datapoint_component:138] | 4. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:42:12 | [W] | [optolink.datapoint_component:138] | 5. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:42:13 | [D] | [text_sensor:064] | 'Component state': Sending state 'OK'
01:42:22 | [W] | [optolink.datapoint_component:138] | 6. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:42:32 | [W] | [optolink.datapoint_component:138] | 7. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:42:33 | [D] | [text_sensor:064] | 'Component state': Sending state 'OK'
01:42:42 | [W] | [optolink.datapoint_component:138] | 8. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:42:52 | [W] | [optolink.datapoint_component:138] | 9. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:42:53 | [D] | [text_sensor:064] | 'Component state': Sending state 'OK'
01:43:02 | [W] | [optolink.datapoint_component:138] | 10. read request for Device Info rejected  due to outstanding running request - increase update_interval!
01:43:12 | [I] | [optolink:052] | requesting value of datapoint Device Info
01:43:12 | [D] | [optolink:082] | VitoWiFi: queue full
01:43:12 | [E] | [optolink:054] | read request rejected due to queue overload - queue size: 2
01:43:12 | [D] | [optolink:056] | queued datapoint: Device Info
01:43:13 | [D] | [text_sensor:064] | 'Component state': Sending state 'OK'
01:43:22 | [I] | [optolink:052] | requesting value of datapoint Device Info
01:43:22 | [D] | [optolink:082] | VitoWiFi: queue full

RevLaw, 2024-04-08:
https://imgur.com/a/Rgn8Nkl
Here is a Video of my adapter hold to the interface, I think the communication should be working or?

RevLaw, 2024-04-08:
I tried the debugging methode where you point your IR onto your Photoresistor: https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702

The Photoresistor is not reacting to the IR diode and I get no output in the log. But when I grab an old IR remote from my TV, the Photoresistor is getting the input and is working fine.

It seems like my IR diode is not working correctly. But it has 1.3V across and shows light when I point my phone camera at it.

I don't know how to troubleshoot further. I don't get why the Photoresistor is not reacting to the wavelength of the IR diode.
Photoresistor: [SFH 309 FA](https://www.reichelt.de/fototransistor-npn-730-1120nm-24-tht-3mm-sfh-309-fa-p60553.html)
IR Diode: [SIR 204 EVL](https://www.reichelt.de/infrarot-diode-gaalas-875-nm-30-3-mm-t1-sir-204-evl-p219694.html)

RevLaw, 2024-04-10:
I now have connected the IR LED directly to ground so it has a persistent output of IR what I checked with the camera. Also, I disconnected the gray cable to the gpio pin. I've attached my multimeter to the PT, and it shows 3.3V. 
If I now point the IR LED on the PT it goes down to like 3.1V but not further. 

Could it be too much resistance for the IR or PT? Or just a broken IR LED?

I don't have an oscilloscope, sorry 😞 

Although yes, the Flashing of the PT was because of stray light from the Vitodes Output and looks the same with the ESP turned off.

rorso, 2024-04-10:
> If I now point the IR LED on the PT it goes down to like 3.1V but not further.

Both the PT and the LED have a small optical angle They should point their headings with good alignment. Distance makes a big different too, so they might directly meet each other to get full brightness. The sensor is designed for less then 5mm distance.

I could not read the values of your resistors from the video. The LED resistor might go as low as 100R for continuous light and in fact it really should be lower than the usual 220R. You are dealing with 3.3V signal not with 5V where the 220R is most widely used. The IR has a little higher forward-voltage either (~1.5V), that you have to subtract from the 3.3V. You may end up with 1.8V remaining on what you dimension the resistor and thus the power.

R = U/I = 1.8V / 0.1A = **18**R for the maximum allowed continuous current of 100mA which is consistent with the maximum 150mW (1.5V * 0.1A) allowed for this diode. If in doubt - lower the LED resistor.

With a signal having short pulses only, you can even go as high as 1A current when I read the data sheet.

You can't easily overload this LED but you can easily dim it too much. If you can't saturate **your** PT with the LED, there is a good chance, the interface will not get switched either. **As long as you see it working with the smartphone cam, it is not broken and powered the right direction.** If you got the right part, then the wavelength is suitable. You would not easily recognize, if not. A 950nm LED looks the same to the cam but not to the PT.

The PT (FA variant) has peak sensitivity at 900nm, the LED at 875nm. Not a perfect match but that's within 80% sensitivity window. And they are not even meant to talk to each other so this should not harm.

The PT behaves similar. The reaction to light within the sensitive wavelength greatly changes by dimming the illumination. Depending on the "group" your PT belongs to, you could end up with a sensitivity difference of up to 700% for your specific sample. The IR LED in the OptoLink is both powerful enough and the proper wavelength - you would not be able to change that on.

A 10K pull-up resistor on the PT is easy to draw down by the PT when illuminated, even for a "group-2" PT. It should go down to (almost) zero. As long as you get it below 1V, it will get counted as "low". There is a chance you just got the PT reversed. The "mark" (flat side) on the case is the collector and to be connected to both the resistor and the input pin. But you cannot break it if you try to reverse it. It just would not switch. 

On the LED the mark is on the cathode. The pins on the "marked" side are the ones directly connected to the I/O ports for both.

I found that I had to LOWER the pull-up of the PT too, to not only get a good high/low value but a fast transition between them. I settled at 5K instead of 10K. The PT must draw more current to switch low - which it does without complaining - but the recovery to the "high" state was MUCH better, giving a much cleaner and square signal and a stable reading of the data. 

On the input of the chip there is a tiny bucket (small capacitance) that has to be filled with electrons before the circuit tells "full = high". These have to pass the pull-up resistor as there is no other source for them. Big resistor value = few electrons = longer time to recover = bad signal edge = bad timing. 10K might be a compromise between low illumination (low down-switching capability) and slow recovery.

Regarding a scope - if you at least occasionally do some electronic projects, you will find that actually looking at the signals might really help diagnosing problems. While a €5.000 Scope certainly is a funny gadget to play with, a €50 pocket scope from ALiExpress will take you a long way.

Talk to those friendly people: https://www.eevblog.com/forum/testgear/portable-oscilloscopes-list/msg5194335/#msg5194335
and try to not acquire [GAS](https://de.wikipedia.org/wiki/Gear_Acquisition_Syndrome).

Hints: 

- Bandwidth rulez! 
- a €25 "toy scope" will easily show you a reasonable waveform for your 4800 bps signal, even if getting a stable trigger for UART data is not easy.
- In the range 25-250€ overall usability grows very quickly. 
- To get a good impression of a 250kHz **digital** signal, you would need x4 = 1MHz bandwidth.

RevLaw, 2024-04-12:
> There is a chance you just got the PT reversed. The "mark" (flat side) on the case is the collector and to be connected to both the resistor and the input pin.

You were so right about that. The problem was just a reversed PT. I have searched for so long, but I am learning a lot from troubleshooting and your expertise. Thank you so much for that.
I googled 'phototransistor direction' and looked at the first picture that came up. Also, I cut the legs as soon as I received them, so I couldn't recognize the polarisation of it on the short and long leg. Now I will never forget it :smile:
Feels bad that I said this:
> The polarisation seems right and all connections on the board are working.

closed 2024-04-12 by RevLaw

rorso, 2024-04-12:
> Now I will never forget it

You memorize those things best, that took a certain amount of effort to achieve. Keep in mind, that every part is different. That's why having the correct datasheet is really important. One has a leg different; one has a marking on the case, ... and there is a chance that this mark means different things to different parts - even with the same manufacturer and device class.

If you look on the datasheet of the "SFH 309 FA" (copied from your link) at page 10, you can see **three** differences on the component that can be used to find the Collector pin:

- A shorter leg
- an oblique angle on the leg
- a flat spot on the rim.

[image]

There is a convention to have the **C**ollector (Transistor) and the **C**athode (LED) marked by a **C**ut leg and a mark on the case. If you have a clear LED you will notice the inner shape of the wires to be different. One is notable bigger and formed like a **C**up. That's the **C**athode again.

But it's just a convention. It might be different on a specific part. 3-color LEDs usually have the Anode formed as a Cup. Datasheet is the only reference.