openv/openv #642: Bestimmt schon oft gefragt, aber ich finde es nicht: Kann jemand mir gegen Aufwandsentschädigung einen USB_Duo_Sniffer bauen?

Gandralf1969, 2024-04-25:
Habe leider eher zwei "linke Hände" und null Erfahrung, was das Selbstzusammenbauen solcher Komponenten angeht.
Trotzdem würde ich gerne die Kommunikation unserer VITOCAL 200S mit dem VitoConnect "mitlesen",
um besser zu verstehen, warum sie selbst im Sommer mehr als 10 mal - kurz - anspringt.

Ja, da sollte mir mein Installateur helfen können, aber der hat die Frage auch nur an VIESSMANN weitergeleitet.
(unbefriedigende) Antwort: Alles normal!  :-/

Um besser zu verstehen, welche Betriebszustände zu diesem - sicher für den Verdichter nicht gesunden! - "Takten" führen,
würde ich die Kommunikation mit dem Vitoconnect gerne mitlesen und auf einem Raspberry 4B protokollieren.

Lange Rede, kurzer Sinn: Wo kann ich den USB_Duo_Sniffer bekommen ... denn alles andere hätte ich wohl schon.  :o)

Die Beschreibungen der Programme (VCONTROLD etc.) sind ja vorhanden, damit sollte ich zurecht kommen
...und falls nicht, dann gibt es ja dieses Forum!  :oD

timbrd, 2024-04-25:
Ich habe genau die gleiche Kombination an Komponenten und wollte auch die Werte auslesen. Schau mal [hier](https://github.com/philippoo66/optolink-splitter), da brauchst du keinen Duo Sniffer, sondern lediglich einen Adapter zwischen Raspi und  der Vitoconnect, den du dir bei Amazon, Reichelt etc. bestellen kannst.

Gandralf1969, 2024-04-26:
Hallo @timbrd 

Vielen Dank für die schnelle Rückmeldung!
Ich hätte jetzt eher so etwas wie ein Splitter- oder Y-Kabel erwartet, also etwas, wo ich das OptoLink-Kabel einstecke und was dann zwei Ausgänge hat (einmal zum Vitoconnect und einmal zum Raspi)
Wie verbinde ich die Komponenten denn in Deiner Lösung?  (Ggf. Bitte ein paar Fotos schicken)

Oder muss ich das OptoLink-Kabel auftrennen um statt des bisherigen USB-Steckers den von Dir verlinkten CP2102-Adapter zu verbinden?

Sorry, wenn ich so laienhaft Frage, aber dies ist wirklich mein erstes "Projekt" dieser Art... :o)

timbrd, 2024-04-26:
Sowas hatte ich auch erst im Sinn, aber die Lösung über den Optolink Splitter ist zumindest hardwaretechnisch einfacher.
Du nimmst das Optolink-Kabel und steckst es in den Pi. Den Adapter schließt du mit den mitgelieferten Jumperkabel wie [hier](https://www.jeffgeerling.com/blog/2021/attaching-raspberry-pis-serial-console-uart-debugging) beschrieben an den Pi an, den USB-Stecker in die Vitoconnect.
Solltest du einen Pi 3 oder 4 haben, dann musst du auf der SD-Karte noch die config.txt bearbeiten und folgendes hinzufügen bzw. die bestehende Zeile entsprechend abändern:

````
dtoverlay=miniuart-bt      # für Pi 4
dtoverlay=pi3-miniuart-bt  # für Pi 3
````

Anschließend den Optolink Splitter auschecken, settings.ini anpassen (insb. ´/dev/ttyAMA0` anstatt `/dev/ttyS0`), die Python Pakete ´paho´ und ´pyserial´ nachinstallieren und zum Testen die optolinkvs2.py starten. Die Status-LED der Vitoconnect sollte dann wieder durchgängig leuchten und damit anzeigen, dass wieder eine Verbindung zur Vitocal und zu Viessmann besteht.

Wenn du Fragen hast, dann stell sie am Besten in den Diskussionen des Github Projektes. Der Entwickler ist sehr hilfsbereit.
Ich kämpfe aktuell mit den richtigen Einstellungen, um die für mich interessanten Werte der Vitocal auszuleiten. Aber das Problem scheinen ja alle zu haben :)

Gandralf1969, 2024-04-26:
Hallo Tim,

Vielen Dank für die ausführliche Anleitung!
Damit sollte selbst ich es schaffen ;o)

Schönes Wochenende!