openv/openv #649: USB0 error

nosektomas2004, 2024-07-04:
Hello.
Thanks a lot for the project, it's great!

I use Vcontrold for HomeAssistant. But I have a problem - probably not much related to this project, but I'll ask, maybe someone will have experience with it.

I have an Optolink USB (post-built from a friend - the base is a USB TTL converter and behind it transistors and IR LEDs etc..). Everything works...but only "for a while". within hours. Always after a few hours of running the plugin crashes because an error occurs on /dev/ttyUSB0.

```
[75] Thu Jul 4 08:13:48 2024 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
[75] Thu Jul 4 08:13:48 2024 : Configuring serial interface /dev/ttyUSB0
[75] Thu Jul 4 08:13:48 2024 : error ioctl TIOCMSET /dev/ttyUSB0:Input/output error
```

When I disconnect and reconnect the USB TTL (and start the add-on), it works again right away for a while. Has anyone encountered this? Does anyone know what causes this?

Thanks a lot

vich-667, 2024-07-05:
I had similar issues. Useing a good USB Cable with less than 1m solved my issue.

nosektomas2004, 2024-07-15:
> I had similar issues. Useing a good USB Cable with less than 1m solved my issue.

Thanks, I'll give it a try

nosektomas2004, 2024-09-03:
Unfortunately, that didn't work. I tried putting a USB hub in front of the module, swapping usb/ttl modules and it didn't help ...

Anyone any advice on what to do?

Thanks a lot!