openv/openv #650: Energy values 0x16A0 to 0x16A5

FransOv, 2024-07-13:
I'm using Optolink-splitter (https://github.com/philippoo66/optolink-splitter) to read and control our Vitocal 300-G with Vitotronic 200 WO1C. I managed to find most of the relevant datapoints by querying the SQL database of the Vitosoft program. However some datapoint addresses in the database are incorrect or have very cryptic descriptions.
What I am looking for now is the actual meaning of the energy readings at 0x16A0 to 0x16A5.  0x16A4 is, I'm fairly sure, the power consumption of the compressor. It is always about 150 to 180 W less then the readings of the external power meter. 0x16A0 is probably the thermal power. 0x16A2 is the unknown. the difference between 0x16A0 and 0x16A2 is clearly correlated with but always less then the reading of 0x16A4.
I assume the odd-numbered addresses are for the second compressor in a tandem configuration. Always zero.
Anyone any ideas? Attached is a graph of the readings. The orange line is of the external power meter.
[image]

budisek, 2025-06-03:
Hey have you figured this out? I have Vitocal 200-S and can't find addresses setup setup to get energy consumption data.

FransOv, 2025-06-03:
See tab  CU401B_S in the Excel at  https://github.com/philippoo66/optolink-splitter/wiki/310-Parameter-Addresses