openv/openv #651: Vitodens200, Vitotronic 200, SM1 - Solar entities?

nosektomas2004, 2024-07-15:
Hi, I have a Vitodens200 with Vitotronic 200 combination connected via optpolink to HA and I am reading out information from it. My deviceID is 20CB. I also have a solar controller SM1 connected to the system for photothermal. Does anyone have the vito.xml configuration for the solar data please? I was able to read the collector temperature but nothing more. I would like to read also:

1) Solar status (icon on the main screen; whether the solar is "heating")
[image]

2) Solar hot water temperature
[image]

3) Solar pump speed
[image]

4) Amount of energy delivered today
[image]

Does anyone please have a similar solar entity that you are reading?
Thanks

nosektomas2004, 2024-07-23:
Anyone with this experience?

I also don't know what the default "getSolarStatus" command means yet. Sometimes it changes from 1 to 0 (when I restart the boiler), but 99% of the time it's 1 (true). I thought that would be what I was trying to read in the case of 1.), but it doesn't look like it... So I don't know exactly what it means.

jaegerschnitzel, 2024-12-14:
Did you find out more in the last months?
I have the same combination and trying to get the data as well.

nosektomas2004, 2025-01-17:
Hi, sorry for the late reply- I didn't notice the announcement. 

I still haven't finished all the entities I'd like, but I think I have SOLAR complete. The only thing is that I don't have the solar pump status as a percentage, but only the 'On/Off' status (corresponds to the solar panel icon displayed on the control). I'm sending my SOLAR configuration section to vito.xml:
```

<!-- .....SOLAR..... -->
    <!-- Teplota kolektoru  -->
    <command name="getTempKol" protocmd="getaddr">
      <addr>6564</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Ermittle die Kollektortemperatur in Grad C</description>
    </command>
    <!-- Provozni hodiny solaru  -->
    <command name="getSolarStunden" protocmd="getaddr">
      <addr>6568</addr>
      <len>2</len>
      <unit>CO</unit>
      <description>Solar Betriebsstunden</description>
    </command>
    <!-- Deni dodana energie -->
    <command name="getSolarLeistungTag" protocmd="getaddr">
      <addr>CF30</addr>
      <len>4</len>
      <unit>CO</unit>
      <description>Solarertrag Aktueller Tag</description>
    </command>
    <!-- Celkova dodana kumulativni energie  -->
    <command name="getSolarLeistung" protocmd="getaddr">
      <addr>6560</addr>
      <len>4</len>
      <unit>CO</unit>
      <description>Solar Leistung Gesamt</description>
    </command>
    <!-- Potlaceni dohrevu TUV -->
    <command name="getSolarStatusWW" protocmd="getaddr">
      <addr>6551</addr>
      <len>1</len>
      <unit>RT</unit>
      <description>Ermittle den Status der Nachladeunterdrueckung</description>
    </command>
    <!-- FTE v provozu - stav solarniho cerpadla -->
    <command name="getPumpeDrehzahlSolar" protocmd="getaddr">
      <addr>6552</addr>
      <len>1</len>
      <unit>RT</unit>
      <description>Ermittle die Drehzahl der Internen Pumpe</description>
    </command>
    <!-- Teplota akumulacka spodni cidlo -->
    <command name="getTempS4" protocmd="getaddr">
      <addr>6566</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Ermittle die Temperatur S4 in Grad C</description>
    </command>
```
I have it with Czech comments (and German ones), so hopefully that will help you. Let me know if it works for you too

Purely by chance - do you have an entity whether the boiler heats on reduced operation or normal (sun or moon on the control)? I can't seem to find this information anywhere...

Thank you!