openv/openv #657: Fehler im Schaltbild

BastiJoe, 2024-10-14:
Hallo
in der Raspberry PI Bauanleitung ist bei der IR LED beschrieben, dass die Anode der kurze Anschluss und die abgeflachte Seite ist.
Es ist aber die Kathode.
https://github.com/openv/openv/wiki/Bauanleitung-RaspberryPi

Viele Grüße

Sebastian

BastiJoe, 2024-10-14:
Tatsache... 
Ich habe die Platine nach dem Schaubild gebaut und die LED hat nicht funktioniert. Ich habe sie umgedreht und dann gings.
"normale" LEDs sind so beschrieben :
[image]

pDR8iLQ4go, 2025-03-16:
Das hat mich auch verwirrt. Hier ein Bild von der LED, dich ich nutze: L-7104SF4BT

[image]

Um ganz sicher zu sein (> 100%) have ich die LED mit einem Vorwiderstand an 5V angeschlossen. Da eine Handy-Kamera auch IR sieht, kann man so überprüfen von + (Anode) bzw. - (Kathode) ist.