openv/openv #658: SFH487-2 Infrarot-LED

Wuifi, 2024-10-17:
Falls Bedarf besteht, ich habe hier noch ca 20 SFH487-2 Infrarot-LED liegen.
Schreibt mir ne PN und ich melde mich!

thomas-omann, 2024-10-19:
Hi Wuifi,

ich bin neu bei Github und weiß noch nicht, wie man ne PN schreiben kann.
Ich hätte Interesse, 5 Stück, Versand via Brief, Zahlung über paypal. Der Gehäuse-Durchmesser beträgt 3mm, oder...?

Beste Grüsse

Thomas

olomberg, 2024-10-20:
Hi Wufi.

Hätte auch Interesse. Melde dich einfach. Dank.

Gruss

Oliver

Ventusfahrer, 2024-10-31:
ich bräuchte 3 Stück, gerne per Brief, Zahlung per PayPal ist ok.

Wäre schöne, wenn das klappen würde!

Danke

Peter

TingelRingel, 2024-11-18:
Hallo,
Falls du noch zwei hast würde ich gerne welche nehmen.

Lg

newkit, 2025-04-20:
Hi @Wuifi,

ich hätte auch interesse falls du noch welche hast.
Zwei Stück per Brief und Paypal?

LG
Nils

TingelRingel, 2025-04-21:
Ich habe seinerzeit keine Antwort bekommen und als Ersatz SIR 204 EVL bestellt. Die sind bei vielen Händlern verfügbar und funktionieren auch einwandfrei.

Lg

newkit, 2025-04-21:
Danke für den Tipp @TingelRingel!