openv/openv #660: ESPHome / Demos D1 mini / Optolink no communication

lsrs7, 2024-10-20:
hi,

ich habe eine Vitowifi auf einer Nodemcu laufen und möchte jetzt auf ESPHome / Optolink umsteigen ([https://github.com/openv/openv/wiki/ESPHome-Optolink]

Die Installation klappt soweit, allerdings fliessen keine Daten. Evtl. sind die Ports für den D1 nicht richtig hinterlegt.

YAML:
```
substitutions:
  name: esphome-web-df9697
  friendly_name: vD200test2

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp8266:
  board: esp01_1m
  
# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://pr#4453
    components: [ optolink ]
    refresh: 0s

optolink:
  protocol: P300             # P300 oder KW
  logger: enable             # wenn später alles rund läuft, wieder entfernen 
  #state: Status              # wenn später alles rund läuft, wieder entfernen
  #device_info: 0x00F8      # dient zur Identifizierung der Vitotronic 

sensor:
  - platform: optolink
    name: ${name} Kesseltemperatur
    address: 0x0810
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature
```

-----------------------------

```
20:18:14	[I]	[optolink:093]	
resuming communication
20:18:14	[I]	[optolink:103]	
requesting value of datapoint esphome-web-df9697 Kesseltemperatur
20:18:14	[D]	[optolink:138]	
VitoWiFi: queue full
20:18:14	[E]	[optolink:107]	
read request rejected due to queue overload - queue size: 2
20:18:14	[W]	[optolink:085]	
communication disrupted - suspending communication for 10 sec; timestamp_loop: 474431, timestamp_send: 0,  timestamp_receive: 0 
20:18:14	[D]	[optolink:110]	
queued datapoint: esphome-web-df9697 Kesseltemperatur
20:18:24	[I]	[optolink:093]	
resuming communication
```

j0ta29, 2024-10-23:
Hallo @lsrs7,
ich vermute, dass Du noch ein Problem mit dem Adapter hast. Der Log-Auszug deutet jedenfalls darauf hin.
Ob die Senderichtung des Adapters funktioniert, kannst Du mit der Handy-Kamera testen. Die macht die IR-Wellenlänge der Sendediode sichtbar.
Der Test der Empfangsrichtung ist leider nicht so einfach. Im PR (https://github.com/esphome/esphome/pull/4453) gibt es einen EspHome-Testkonfiguration (Eintrag von derwassi am 23.5.2023), die den Adapter testet.

Das Log einer funktionierenden Installation sieht z.B. so aus:
```
[15:53:06][I][optolink:103]: requesting value of datapoint Gastherme Niveau Heizkennlinie
[15:53:06][D][optolink:138]: VitoWiFi: READ Gastherme Niveau Heizkennlinie
[15:53:06][D][optolink:138]: VitoWiFi: READ 4105000127d40102
[15:53:06][D][optolink:138]: VitoWiFi: ack
[15:53:08][D][optolink:138]: VitoWiFi: RCV 4106010127d4010004
[15:53:08][D][optolink:138]: VitoWiFi: ack
[15:53:08][D][optolink:138]: VitoWiFi: DP Gastherme Niveau Heizkennlinie succes
[15:53:08][I][optolink.datapoint_component:037]: recieved data for datapoint Gastherme Niveau Heizkennlinie: 0
```
Mich wundert auch, dass bei Dir die Einträge von VitoWiFi fehlen. Eventuell kommen die weiter vorne im Log.

lsrs7, 2024-10-27:
@j0ta29,

danke für die Tipps. Ich habe in der Tat die LED ausgetauscht und sie blinkt jetzt im Kamera-Bild. Den Eintrag von für die Testkonfiguration habe ich nicht gefunden.

j0ta29, 2024-10-29:
Hier ist der direkte Link zur Testkonfiguration: https://github.com/esphome/esphome/pull/4453#issuecomment-1559988702

lsrs7, 2024-10-31:
@j0ta29,
that example I've found, but it is not from user "derwassi" as mentioned. I used that code, but no communication is established :/ I've recorded some graphs of the led and transistor and it looks like that the low's are to high. 

[image]
[image]

j0ta29, 2024-11-01:
Your configuration already is fine.
The log messages from the VitoWifi library unfortunately aren't shown in the HTML user interface of EspHome. You only see black lines.
Use the esphome command line tool to see the messages.

lsrs7, 2024-11-01:
Could you give me some hints to use the esp home cli to retrieve the logs?

lsrs7, 2024-11-03:
You can close, doesn't work on Wemos D1 mini, switched to ESP32 and all works.

j0ta29, 2024-11-04:
Did you find out why it doesn't work with D1 mini?

j0ta29, 2024-11-04:
By the way: information about the EspHome command line interface can be found here:
- https://esphome.io/guides/installing_esphome
- https://esphome.io/guides/cli.html

LPTT, 2024-11-07:
Installed it today on a d1 mini. It works!

j0ta29, 2024-11-07:
@LPTT , are you using a Wemos D1 mini. I'm using this board and have no problems at all (board: d1_mini)
The board of @lsrs7 probably is a variant of the Wemos board.

lsrs7, 2024-11-10:
That might be the reason, it looks like that the rx pin is behaving differently.

darth-hp, 2025-03-02:
```yaml
binary_sensor:
   - platform: gpio
     pin: 
       number: GPIO20
     name: "test"
     on_click: 
       then:
         - logger.log: "received 2"
     on_press: 
       then:
         - logger.log: "received"
```
Hallo zusammen,

Mein dem obigen Snippet habe ich festgestellt, dass Sender und Empfänger direkt miteinander kommunizieren können da keine ausreichende Abschirmung zwischen den beiden war, was natürlich zu Fehlern führt. Mit etwas anderem Setup konnte ich das Problem entfernen. Das schalten hat einwandfrei und wie erwartet funktioniert, doch leider konnte ich bisher noch keinerlei Daten empfangen.

In meinem Setup habe ich diese IR Module verbaut https://de.aliexpress.com/item/1005004005620672.html - ist es wichtig welche IR Dioden man verwendet?

j0ta29, 2025-03-06:
Mit meinen bescheidenen Eletronikkenntnissen würde ich behaupten, dass zwei Dinge wichtig sind:
a) Wellelänge der IR-Diode.
b) Vorwiderstand

darth-hp, 2025-03-06:
Da ja die Kommunikation funktioniert, gehe ich von a) aus. Definitiv passende Dioden kommen bald.

darth-hp, 2025-04-03:
Passende Dioden installiert - jetzt funktioniert es.

darth-hp, 2025-04-05:
Ich packe noch eins drauf - die Abschirmung der Dioden ist extrem wichtig. Wenn die sich gegenseitig "sehen" kommt Müll raus.