openv/openv #661: ESPHome Optolink communication disrupted or API connection resets

olyphar, 2024-12-05:
Hi all,

I am looking for support regarding my trouble with the [ESPHome Optolink ](https://github.com/openv/openv/wiki/ESPHome-Optolink) component by @j0ta29.

First of all I would like to mention that I got [Vitoconnect for ESPHome](https://github.com/dannerph/esphome_vitoconnect/tree/master) by @dannerph running successfully, logs and configuration attached, see
[logs_esphome-web-48c66c_run (4).txt](https://github.com/user-attachments/files/18021343/logs_esphome-web-48c66c_run.4.txt) and 
[esphome-vitoconnect2-nopwd.yaml.txt](https://github.com/user-attachments/files/18021374/esphome-vitoconnect2-nopwd.yaml.txt)

Once I change my configuration to ESPHome Optolink (OTA), I am getting two different types of errors. I have already tried different UART for the logger or completely disabling logger at all or setting baud rate to 0 instead, with no improvements whatsoever.

Any guesses or ideas what I could try are welcome.
My hardware is an ESP32WROOM dev board.

1.  `[22:06:08][W][optolink:088]: communication disrupted - suspending communication for 10 sec; timestamp_loop: 56848, timestamp_send: 54843,  timestamp_receive: 0 ` - and communication never recovers afterwards

```
[22:05:36][I][optolink.datapoint_component:055]: recieved data for datapoint Device Info: 721472288
[22:05:36][D][text_sensor:064]: 'Device Info': Sending state 'Device ID: 20CB|Hardware Revision: 00|Software Index: 2B'
[22:05:36][W][component:237]: Component optolink took a long time for an operation (305 ms).
[22:05:37][W][component:238]: Components should block for at most 30 ms.
[22:05:37][I][optolink.datapoint_component:079]: recieved data for datapoint Kesseltemperatur Tiefpass: 56.000000
[22:05:37][D][sensor:094]: 'Kesseltemperatur Tiefpass': Sending state 56.00000 °C with 0 decimals of accuracy
[22:05:38][W][component:237]: Component optolink took a long time for an operation (304 ms).
[22:05:38][W][component:238]: Components should block for at most 30 ms.
[22:05:38][I][optolink:103]: requesting value of datapoint Kesseltemperatur Tiefpass
[22:05:46][I][optolink:103]: requesting value of datapoint Device Info
[22:05:48][W][optolink.datapoint_component:148]: 2. read request for Kesseltemperatur Tiefpass rejected due to outstanding running request - increase update_interval!
[22:05:48][W][component:237]: Component optolink.sensor took a long time for an operation (295 ms).
[22:05:48][W][component:238]: Components should block for at most 30 ms.
[22:05:49][W][optolink:088]: communication disrupted - suspending communication for 10 sec; timestamp_loop: 37646, timestamp_send: 35330,  timestamp_receive: 26425 
[22:05:49][W][component:237]: Component optolink took a long time for an operation (593 ms).
[22:05:49][W][component:238]: Components should block for at most 30 ms.
[22:05:58][I][optolink:093]: resuming communication
[22:06:06][I][optolink:103]: requesting value of datapoint Gerätekennung
[22:06:08][W][optolink:088]: communication disrupted - suspending communication for 10 sec; timestamp_loop: 56848, timestamp_send: 54843,  timestamp_receive: 0 
[22:06:08][W][component:237]: Component optolink took a long time for an operation (295 ms).
[22:06:08][W][component:238]: Components should block for at most 30 ms.
[22:06:09][D][text_sensor:064]: 'Component state': Sending state 'communication suspended'

```

2. API communication resets

```
[22:02:44][I][optolink.datapoint_component:079]: recieved data for datapoint Kesseltemperatur Tiefpass: 53.299999
[22:02:44][D][sensor:094]: 'Kesseltemperatur Tiefpass': Sending state 53.30000 °C with 0 decimals of accuracy
[22:02:44][C][safe_mode:018]: Safe Mode:
[22:02:44][C][safe_mode:020]:   Boot considered successful after 60 seconds
[22:02:44][C][safe_mode:021]:   Invoke after 10 boot attempts
[22:02:44][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[22:02:44][C][api:140]: API Server:
[22:02:44][C][api:141]:   Address: viessmann-reader.local:6053
[22:02:44][C][api:145]:   Using noise encryption: NO
[22:02:54][I][optolink:103]: requesting value of datapoint Kesseltemperatur Tiefpass
[22:02:54][I][optolink.datapoint_component:079]: recieved data for datapoint Kesseltemperatur Tiefpass: 53.299999
[22:02:54][D][sensor:094]: 'Kesseltemperatur Tiefpass': Sending state 53.30000 °C with 0 decimals of accuracy
[22:02:55][D][api:103]: Accepted 192.168.11.11
[22:02:55][D][api.connection:1446]: Home Assistant 2024.11.3 (192.168.11.11): Connected successfully
[22:02:56][W][api.connection:105]: Home Assistant 2024.11.3 (192.168.11.11): Connection closed
[22:02:56][D][api:103]: Accepted 192.168.11.11
[22:02:56][D][api.connection:1446]: Home Assistant 2024.11.3 (192.168.11.11): Connected successfully
[22:02:56][W][api.connection:105]: Home Assistant 2024.11.3 (192.168.11.11): Connection closed
[22:02:56][D][api:103]: Accepted 192.168.11.11
[22:02:56][D][api.connection:1446]: Home Assistant 2024.11.3 (192.168.11.11): Connected successfully
[22:02:56][W][api.connection:105]: Home Assistant 2024.11.3 (192.168.11.11): Connection closed
[22:02:56][D][api:103]: Accepted 192.168.11.11

``` 

YAML:

```
external_components:
  - source: github://pr#4453
    components: [ optolink ]

esphome:
  name: viessmann-reader
  friendly_name: vitoconnect

esp32:
  board: esp32dev

api:

logger:
  hardware_uart: UART1
  #baud_rate: 0
  #level: DEBUG

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Openv Fallback Hotspot"
    password: "0IHlfkae8cGq"

optolink:
  rx_pin: GPIO16              # ESP32 RX2
  tx_pin: GPIO17              # ESP32 TX2
  protocol: P300             # P300 oder KW
  logger: enable             # wenn später alles rund läuft, wieder entfernen 

sensor:
  - platform: optolink
    name: "Kesseltemperatur Tiefpass"
    address: 0x0810
    bytes: 2
    div_ratio: 10
    unit_of_measurement: "°C"
    device_class: temperature
    update_interval: 10s

number:
  - platform: optolink
    name: "Sollwert Raumtemperatur"
    unit_of_measurement: °C
    address: 0x2306
    bytes: 1
    min_value: 3
    max_value: 37
    step: 1
    mode: box
    icon: "mdi:home-thermometer"
    device_class: temperature
    update_interval: 120s

text_sensor:
  - platform: optolink
    name: "Heating operation mode"
    address: 0x2323
    bytes: 1
    update_interval: 300s
    type: MAP
    filters:
      map:
        - "0 -> Off"
        - "1 -> Only hot water"
        - "2 -> Heating and hot water"
  - platform: optolink
    type: DEVICE_INFO
    name: "Device Info"
    update_interval: 60s
  - platform: optolink
    type: STATE_INFO
    name: "Component state"
    update_interval: 60s
  - platform: optolink
    type: RAW
    name: "Gerätekennung"
    address: 0x00F8
    bytes: 2
    update_interval: 60s
```

j0ta29, 2024-12-13:
@olyphar , I don't know the implementation of Vitoconnect for ESPHome, but I think the last line in the log of Vitoconnect also looks like a problem with the harware/interface.

In the logs of ESPHome Optolink I can see that there are successful reads of data from the interface. Unfortunately not for a long time.

I recommend the following logging config...
```
logger:
  hardware_uart: UART1
  baud_rate: 0
  level: DEBUG

optolink:
  rx_pin: GPIO16              # ESP32 RX2
  tx_pin: GPIO17              # ESP32 TX2
  protocol: P300             # P300 oder KW
  logger: true             # wenn später alles rund läuft, wieder entfernen 
```

With the command line interface of EspHome (https://esphome.io/guides/cli.html) you should see low level debug information that could help for further investigation of your problem.

olyphar, 2024-12-13:
Thanks for your advice, @j0ta29!
I have now a) used your proposed logging configuration and b) logged everything with CLI with "verbose" option.

Even though the Vitoconnect for ESPHome component shows an error, it continuously delivers updated (and correct) values to Home Assistant.

At the same time, the Optolink component seems to be reading correctly (no "communication disrupted" with these settings) but API connection seems to be reset. Can I somehow get even more details on _why_ the connection is closed?

'''[15:40:27][D][api:103]: Accepted 192.168.11.11
DEBUG viessmann-reader @ 192.168.11.115: Got message of type SubscribeLogsResponse: level: LOG_LEVEL_DEBUG
message: "\033[0;36m[D][api.connection:1446]: Home Assistant 2024.12.2 (192.168.11.11): Connected successfully\033[0m"

[15:40:27][D][api.connection:1446]: Home Assistant 2024.12.2 (192.168.11.11): Connected successfully
DEBUG viessmann-reader @ 192.168.11.115: Got message of type SubscribeLogsResponse: level: LOG_LEVEL_WARN
message: "\033[0;33m[W][api.connection:105]: Home Assistant 2024.12.2 (192.168.11.11): Connection closed\033[0m"

[15:40:27][W][api.connection:105]: Home Assistant 2024.12.2 (192.168.11.11): Connection closed
DEBUG viessmann-reader @ 192.168.11.115: Got message of type SubscribeLogsResponse: level: LOG_LEVEL_DEBUG
message: "\033[0;36m[D][api:103]: Accepted 192.168.11.11\033[0m"

[15:40:27][D][api:103]: Accepted 192.168.11.11
DEBUG viessmann-reader @ 192.168.11.115: Got message of type SubscribeLogsResponse: level: LOG_LEVEL_DEBUG
message: "\033[0;36m[D][api.connection:1446]: Home Assistant 2024.12.2 (192.168.11.11): Connected successfully\033[0m"

[15:40:27][D][api.connection:1446]: Home Assistant 2024.12.2 (192.168.11.11): Connected successfully
DEBUG viessmann-reader @ 192.168.11.115: Got message of type SubscribeLogsResponse: level: LOG_LEVEL_WARN
message: "\033[0;33m[W][api.connection:105]: Home Assistant 2024.12.2 (192.168.11.11): Connection closed\033[0m"

[15:40:27][W][api.connection:105]: Home Assistant 2024.12.2 (192.168.11.11): Connection closed
DEBUG viessmann-reader @ 192.168.11.115: Got message of type SubscribeLogsResponse: level: LOG_LEVEL_DEBUG
message: "\033[0;36m[D][api:103]: Accepted 192.168.11.11\033[0m"

'''

The consequences in Home Assistant look like this: Once I switch to Optolink, I have interrupted measurements, as soon as I switch back, to Vitoconnect, they're looking good. 
Why not just keep using Vitoconnect? Because Optolink as way more features implemented like setting values....
[image]

j0ta29, 2024-12-24:
@olyphar, could you please send me a **short** log snippet both from ESPHome log and Home Assistant log with a sequence of disconnect and reconnect action. In Home Assistant at the page Integration->ESPHome you find an option "Enable debug logging" that activates extra information.

olyphar, 2025-07-21:
Hey @j0ta29, since it took me months to get back to smart home stuff, the issue has resolved itself with some other update apparently. After everything is up to date, also the ESPHome Optolink component is running fine. Thanks a lot!

closed 2025-07-21 by olyphar