openv/openv #671: Frontend Auswertung von VcontrolD-Daten

bibcoder, 2025-02-09:
Hallo Leute,

ich werte jetzt die Temperatur-Daten und Solldaten meiner Viessmann Vitoladens 300C (verbunden über Optolink) an meinen MiniPC alle 5 Minuten aus und schreibe sie in einen MySQL-Speicher. Das klappt jetzt wunderbar, vielen Dank allen für diese tolle Arbeit, insbesonderen am Wiki hier und an dem Programm VControlD.

Jetzt frage ich mich, wie es weitergehen soll, und bin ein wenig überfordert. Ums ein wenig konkreter zu sagen, wären das meine Fragen:

1.) Welches Frontend würde sich denn aktuell so für einen Linux-Server anbieten? 
2.) Worauf sollte man bei diesen Auswertungskurven achten, um eventuell Einsparpotential zu entdecken?
3.) Was habt ihr als nächsten Schritt so gemacht?

Schönen Sonntag
cody

MarenB, 2025-02-27:
Moin,

ich setze seit ein paar Jahren Volkszähler auf einem Pi ein, würde aber heute eher Grafana empfehlen. Eventuell wäre es sinnvoll, von MySQL auf influxDB zu wechseln, aber das kann ich nicht mit Sicherheit sagen.

bibcoder, 2025-07-07:
Vielen dank. Dann probier ich auch mal Grafana aus... damit hab ich aber noch nie was gemacht :-O