openv/openv #674: Failed Config

Folllast, 2025-02-27:
`external_components:
  - source: github://pr#4453
    components: [ optolink ]

esphome:
  name: heizung
  friendly_name: Heizung

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heizung Fallback Hotspot"
    password: "y0qjHcrswBaF"

  
optolink:
  protocol: P300             # P300 oder KW
  logger: enable             # wenn später alles rund läuft, wieder entfernen 
  state: Status              # wenn später alles rund läuft, wieder entfernen
  device_info: Gerätekennung # dient zur Identifizierung der Vitotronic 

sensor:
  - platform: optolink
    name: Heizung Kesseltemperatur
    address: 0x0810
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature

captive_portal:
    `

Hello, this Code lead to this error:

`Failed config

optolink: [source /config/esphome/heizung.yaml:41]
  protocol: P300
  logger: enable
  
  [state] is an invalid option for [optolink]. Please check the indentation.
  state: Status
  
  [device_info] is an invalid option for [optolink]. Please check the indentation.
  device_info: Gerätekennung`

Where is my mistake?

j0ta29, 2025-02-27:
The attributes 'state' and 'device_info' are deprecated. Please replace it with:

```
text_sensor:
  - platform: optolink
    type: DEVICE_INFO
    name: ${name} Gerätekennung
    update_interval: 600s
  - platform: optolink
    type: STATE_INFO
    name: ${name} Gerätestatus
    update_interval: 1s
```

Documentation for the current development can be found here:
https://deploy-preview-2737--esphome.netlify.app/components/optolink

j0ta29, 2025-03-05:
Framework `arduino` is mandatory for the optolink component.
If the error `MissingPackageManifestError` still exists with framework `arduino` please send your full configuration.