openv/openv #675: Umstieg von VitoWIFI auf ESPHome - Langsame Kommunikation

DominikWild, 2025-03-02:
Hallo zusammen,

ich habe meine Heizung erfolgreich mit VitoWifi und einem selbstgebauten Optolink Adapter auf einem ESP32 ausgelesenen. Die benötigten Variablen der Heizung habe ich alle 20s abgefragt und es lief problemlos.

Jetzt habe ich den selben Adapter an ein ESP32 mit ESPhome angeschlossen. Die Datenpunkte sind identisch. Irgendwie ist jedoch die Kommunikation  zwischen ESPHome und der Heizung deutlich langsamer und ich erhalte bei einem Abfrageintervall von 20s folgende Fehlermeldungen:

```
[15:10:33][E][optolink:109]: read request rejected due to queue overload - queue size: 24
[15:10:33][D][optolink:111]: queued datapoint: Außentemperatur
[15:10:33][D][optolink:111]: queued datapoint: Außentemperatur gedämpft
[15:10:33][D][optolink:111]: queued datapoint: Kesseltemperatur
[15:10:33][D][optolink:111]: queued datapoint: Speichertemperatur
[15:10:33][D][optolink:111]: queued datapoint: Kesselsolltemperatur
[15:10:33][D][optolink:111]: queued datapoint: Brennerstarts
[15:10:33][D][optolink:111]: queued datapoint: Betriebsstunden Brenner
[15:10:33][D][optolink:111]: queued datapoint: Sammelstörung
[15:10:33][D][optolink:111]: queued datapoint: Brennerstörung
[15:10:33][D][optolink:111]: queued datapoint: Heizkreispumpe
[15:10:33][D][optolink:111]: queued datapoint: Speicherladepumpe
[15:10:33][D][optolink:111]: queued datapoint: Zirkulationspumpe
```
Werte kommen an. Ich habe schon Testweite ein paar Datenpunkte rausgenommen, aber bei einem Abfrageintervall von 20s kalppt es nur mit 3-4 Datenpunkten.

Hat jeamand eine Idee, Warum die Kommunikation so langsam ist und warum es mit VitoWIFI und dem selben Adapter keine Probleme (bei mehr Datenpunkten welche abgefragt werden) gibt?

Anbei mein YAMEL:
```

external_components:
  - source: github://pr#4453
    components: [ optolink ]

esphome:
  name: esphome-web-56d17c
  #name: heizungssteuerung
  friendly_name: Heizungssteuerung
  min_version: 2024.10.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: arduino
    

# Enable logging
#logger:
#  hardware_uart: UART1
#  level: DEBUG
logger:
  hardware_uart: UART0
  baud_rate: 0

# Enable Home Assistant API
api:
   encryption:
    key: "d4vs/nj5G70zIPwbGwpDLlgXBoFqK6ixprsBBz5rwgg="
  #password: "1234"

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: "FRITZ!Box 7490 2,4-GHz"
  password: "45728585280608148543"

# Example configuration entry
#switch:
#  - platform: gpio
#    pin: GPIO02
#    name: "Test Schalter"

# Hardware Control
i2c:
  sda: GPIO21 
  scl: GPIO22

ads1115:
  - address: 0x48

optolink:
  protocol: KW             # P300 oder KW
  #logger: enable             # wenn später alles rund läuft, wieder entfernen 
  #state: Status              # wenn später alles rund läuft, wieder entfernen
  #device_info: Gerätekennung # dient zur Identifizierung der Vitotronic 
  rx_pin: GPIO03
  tx_pin: GPIO01

output:
  - platform: gpio
    pin: 18
    id: 'relay1'
  - platform: gpio
    pin: 19
    id: 'relay2'
    
sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    name: "Stand Heizöltank"
    gain: 6.144
    update_interval: 250s
    accuracy_decimals: 1
    unit_of_measurement: "l"
    filters:
     - median:
        window_size: 240  #240 = 60s
        send_every: 240   #240 = 60s
        send_first_at: 40 #Erster Wert nach 10s
     - calibrate_linear:
         method: least_squares
         datapoints:
          # Map 0.0 (from sensor) to 1.0 (true value)
          - 1.0 -> 0.0
          - 5.0 -> 6000.0
     - round: 1
      
  - platform: optolink
    name: "Außentemperatur"
    address: 0x5525             
    bytes: 2                   
    div_ratio: 10
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    update_interval: 20s

  - platform: optolink
    name: "Außentemperatur gedämpft"
    address: 0x5527             
    bytes: 2                   
    div_ratio: 10
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    update_interval: 20s

 
  - platform: optolink
    name: "Kesseltemperatur"
    address: 0x0810             
    bytes: 2                   
    div_ratio: 10
    unit_of_measurement: "°C"
    accuracy_decimals: 1   
    update_interval: 20s
     
  - platform: optolink
    name: "Speichertemperatur"
    address: 0x0812             
    bytes: 2                   
    div_ratio: 10
    unit_of_measurement: "°C"
    accuracy_decimals: 1  
    update_interval: 20s
         
  - platform: optolink
    name: "Kesselsolltemperatur"
    address: 0x5502             
    bytes: 2                   
    div_ratio: 10
    unit_of_measurement: "°C"
    accuracy_decimals: 1 
    update_interval: 20s 

  - platform: optolink
    name: "Brennerstarts"
    address: 0x088A             
    bytes: 4                   
    div_ratio: 1
    #unit_of_measurement: ""
    accuracy_decimals: 0 
    update_interval: 20s

  - platform: optolink
    name: "Betriebsstunden Brenner"
    address: 0x08A7             
    bytes: 4                   
    div_ratio: 1
    unit_of_measurement: "h"
    accuracy_decimals: 2 
    update_interval: 20s
    filters:
    - lambda: return x / 3600;
    - round: 2 

number:
  #- platform: optolink
  #  name: "Sollwert Raumtemperatur"
  #  unit_of_measurement: °C
  #  address: 0x2306
  #  bytes: 1
  #  min_value: 3
  #  max_value: 37
  #  step: 1
  #  update_interval: 120s

  #- platform: optolink
  #  name: "Sollwert reduzierte Raumtemperatur"
  #  unit_of_measurement: °C
  #  address: 0x2307
  #  bytes: 1
  #  min_value: 3
  #  max_value: 37
  #  step: 1   
  #  update_interval: 120s

  #- platform: optolink
  #  name: "Sollwert  Raumtemperatur Partybetrieb"
  #  unit_of_measurement: °C
  #  address: 0x2308
  #  bytes: 1
  #  min_value: 3
  #  max_value: 37
  #  step: 1 
  #  update_interval: 120s 

  #- platform: optolink
  #  name: "Heizkennlinie Niveaue"
  #  #unit_of_measurement:
  #  address: 0x2304
  #  bytes: 1
  #  min_value: 0
  #  max_value: 100
  #  step: 1 
  #  update_interval: 120s

  #- platform: optolink
  #  name: "Heizkennlinie Steigung"
  #  #unit_of_measurement:
  #  address: 0x2305
  #  bytes: 1
  #  min_value: 0
  #  max_value: 10
  #  step: 1 
  #  div_ratio: 10
  #  update_interval: 120s

binary_sensor:
  - platform: optolink
    name: "Sammelstörung"
    address: 0x0847
    update_interval: 20s

  - platform: optolink
    name: "Brennerstörung"
    address: 0x0883
    update_interval: 20s

  - platform: optolink
    name: "Heizkreispumpe"
    address: 0x2906
    update_interval: 20s

  - platform: optolink
    name: "Speicherladepumpe"
    address: 0x0847
    update_interval: 20s

  - platform: optolink
    name: "Zirkulationspumpe"
    address: 0x0883
    update_interval: 20s

switch:
  #- platform: optolink
  #  name: "Sparbetrieb"
  #  address: 0x2302
  #  update_interval: 120s

  #- platform: optolink
  #  name: "Partybetrieb"
  #  address: 0x02303
  #  update_interval: 120s

  - platform: output
    output: relay1
    name: "Umschalter Optolink"
    on_turn_on:
      then: 
        - output.turn_on: relay2
    on_turn_off:
      then:
        - output.turn_off: relay2
    

text_sensor:
  #- platform: optolink
  #  name: "Betriebsart"
  #  address: 0x2301
  #  bytes: 1
  #  type: MAP
  #  filters:
  #    map:
  #      - "0 -> Warmwasser"
  #      - "3 -> Heizen und Warmwasser"
  #      - "5 -> Abschaltbetrieb"
  #  update_interval: 120s

```

dumpfheimer, 2025-03-03:
@DominikWild du solltest dein wlan passwort entfernen. lg

DominikWild, 2025-03-04:
Hat jemand eine Idee, warum die Kommunikation zwischen Heizung und ESPHome so langsam ist?

j0ta29, 2025-03-06:
@DominikWild , die Meldung `read request rejected due to queue overload` deutet darauf hin, dass vorherige Anfragen an die Heizung nicht beantwortet wurden und es deshalb zu einem "Stau" der Anfragen kommt.
Die eigentliche Schnittstelle von ESPHome optolink basiert auf VitoWiFi, d.h. das Low-Level-Protokoll zwischen Microcontroller und Heizung ist dasselbe.

Mit dem Eintrag 
```
optolink:
  logger: enable             # wenn später alles rund läuft, wieder entfernen 
```
kannst Du übrigens die Kommunikation zwischen Controller und Heizung auf VitoWiFi-Ebene verfolgen und bei Bedarf mit Deiner ursprünglichen Lösung vergleichen.

ESPHome muss sich auch noch um ein paar andere Dingen kümmern, z.B. Kommunikation mit Home-Assistant und andere Dingen, die im der EspHome-Konfiguration festgelegt sind, was dem Microcontroller zusätzliche Arbeit auflastet. Bei Dir z.B. Ansteuerung des I2C-Busses. 
Testweise könntest Du ja mal die I2C-Komponenten herausnehmen.
Wenn das auch nicht hilft, wirf mal einen genaueren Blick auf die Log-Ausgabe, ob dort Anfragen an die Heizung nicht beantwortet werden, oder sehr lange benötigen, bis sie beantwortet werden. 
In meiner Konfiguration habe ich auf einem schwächeren ESP8266 deutlich mehr Sensoren ohne Probleme konfiguriert.

DominikWild, 2025-03-07:
@j0ta29 würdest Du mir deinen YAMEL zeigen? Welchen Abfrageinterval nutzt Du?

```
**_[14:33:21][W][component:237]: Component optolink took a long time for an operation (55 ms).
[14:33:21][W][component:238]: Components should block for at most 30 ms._**
[14:33:22][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=694023 (now=714024)
[14:33:22][I][optolink:105]: requesting value (1 bytes) from datapoint Sammelstörung

[14:33:23][VV][api.service:697]: on_ping_request: PingRequest {}
[14:33:23][VV][api.service:043]: send_ping_response: PingResponse {}
[14:33:23][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=695188 (now=715189)
[14:33:23][I][optolink:105]: requesting value (2 bytes) from datapoint Kesseltemperatur
```

```
[14:33:36][I][optolink.datapoint_component:034]: recieved data for datapoint Speicherladepumpe: 0
[14:33:37][VV][api.service:697]: on_ping_request: PingRequest {}
[14:33:37][VV][api.service:043]: send_ping_response: PingResponse {}
[14:33:38][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=710257 (now=730259)
[14:33:38][I][optolink:105]: requesting value (4 bytes) from datapoint Betriebsstunden Brenner

[14:33:38][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=710584 (now=730589)
[14:33:38][I][optolink:105]: requesting value (2 bytes) from datapoint Außentemperatur gedämpft

[14:33:38][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=710720 (now=730721)
[14:33:38][I][optolink:105]: requesting value (1 bytes) from datapoint Heizkreispumpe
```

```
[14:38:18][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=990257 (now=1010257)
[14:38:18][I][optolink:105]: requesting value (4 bytes) from datapoint Betriebsstunden Brenner

[14:38:18][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=990584 (now=1010585)
[14:38:18][I][optolink:105]: requesting value (2 bytes) from datapoint Außentemperatur gedämpft

[14:38:18][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=990720 (now=1010727)
[14:38:18][I][optolink:105]: requesting value (1 bytes) from datapoint Heizkreispumpe

[14:38:20][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=992534 (now=1012535)
[14:38:20][I][optolink:105]: requesting value (4 bytes) from datapoint Brennerstarts

[14:38:20][VV][scheduler:226]: Running interval 'update' with interval=20000 last_execution=992916 (now=1012917)
[14:38:20][I][optolink:105]: requesting value (1 bytes) from datapoint Zirkulationspumpe
```

Ich habe den Eindruck das der ESP mehrere Anfragen direkt hintereinander schickt, ohne auf Antwort zu warten.

j0ta29, 2025-03-15:
@DominikWild 

_> Ich habe den Eindruck das der ESP mehrere Anfragen direkt hintereinander schickt, ohne auf Antwort zu warten_

Vielleicht ein paar Erläuterungen:
Die Nachricht 'requesting value ... from datapoint' erscheint dann, wenn sie laut den Regeln in deiner Konfiguration (update_interval) benötigt wird. Intern wird sie aber noch nicht an die Therme übermittelt, sondern erst dann, wenn die vorherigen Anfragen beantwortet wurden. Solange das nicht passiert ist, wird die Anfrage erst mal in einer Warteschlange platziert, bis sie dran ist. Allerdings hat die Warteschlange eine feste Größe. Und wenn die voll ist, werden allen weiteren Anfragen gar nicht mehr angenommen. Das heißt aber auch, dass eine Anfrage, die von der Therme nicht richtig verarbeitet werden kann und deswegen die Antwort von der Therme nicht kommt, die ganze Warteschlange immer länger werden lässt. 
Ich würde Dir empfehlen, jeden Datenpunkt in deiner Konfiguration **einzeln** zu prüfen, ob er in der Kategorie ist.
Am besten schaltest Du dann auch das erweiterte Logging (von VitoWifi) temporär ein:

optolink:
  logger: enable             # wenn später alles rund läuft, wieder entfernen 

Was mir noch auffällt: Wenn Deine Therme nicht uralt ist, könntest Du mal auf das Protokoll "P300" wechseln.

Falls es Dir hilft, hier meine Konfiguration (nicht vollständig, aber mit den relevanten Teilen): 

[optolink.yaml.txt](https://github.com/user-attachments/files/19260746/optolink.yaml.txt)

j0ta29, 2025-03-16:
@DominikWild, meine Therme antwortet innerhalb von Bruchteilen einer Sekunde auf Anfragen. Hier ein kurzer Auschnitt aus meinem  Protokoll:
```
[11:56:07][I][optolink:105]: requesting value (1 bytes) from datapoint Gastherme Partybetrieb
[11:56:07][D][optolink:139]: VitoWiFi: READ Gastherme Partybetrieb
[11:56:07][D][optolink:139]: VitoWiFi: READ 410500012303012d
[11:56:07][D][sensor:093]: 'Gastherme WiFi Signal': Sending state -74.00000 dBm with 0 decimals of accuracy
[11:56:07][D][optolink:139]: VitoWiFi: ack
[11:56:07][D][optolink:139]: VitoWiFi: RCV 41060101230301002f
[11:56:07][D][optolink:139]: VitoWiFi: ack
[11:56:07][D][optolink:139]: VitoWiFi: DP Gastherme Partybetrieb succes
[11:56:07][I][optolink.datapoint_component:034]: recieved data for datapoint Gastherme Partybetrieb: 0
``` 
Ich sehe in deinem Protokoll nicht die Ausgaben von VitoWiFi. Ich vermute Du verwendest das Web-Interface von ESPHome?
Wenn die ESPHome CLI verwendest, solltest Du auch die VitoWiFi sehen können. Damit kannst Du dann die genauen Antwortzeiten deiner Therme feststellen.