openv/openv #676: Find out which heater I have

KLelong, 2025-03-06:
Hi,
I would like to know which type of heater I have.
On the type-plate I see VITODENS 300-W
Next to the serial-number I see WB3C
I downloaded the files from philipp0066 (optolink-splitter) and looked in the DP Listen and DP_listen_2 zip-files and I find nothing that seems to apply to my heater.

I got some information from the boiler-interface, using VitoWifi :
I sent this :
```
VitoWiFi::Datapoint("deviceID1", 0x00F8, 4, VitoWiFi::noconv),
VitoWiFi::Datapoint("deviceID2", 0x00FC, 4, VitoWiFi::noconv),
VitoWiFi::Datapoint("outsidetemp", 0x5525, 2, VitoWiFi::div10),
VitoWiFi::Datapoint("boilertemp", 0x0810, 2, VitoWiFi::div10),
VitoWiFi::Datapoint("pump", 0x2906, 1, VitoWiFi::noconv)
```
and I got this back :
```
Raw data received: 20 cb 00 0c
deviceID1: 0c00cb20
Raw data received: 00 00 02 00
deviceID2: 00020000
Raw data received: b6 00
outsidetemp: 18.2
Raw data received: eb 00
boilertemp: 23.5
Raw data received: 01
pump: ON
```
The deviceID's are decode to a uint32_t, so I believe there is an endian-issue.
The other data seems valid, but I would like more information from the heater, eventually also control it.

Can anyone help ?
TIA

pDR8iLQ4go, 2025-03-16:
Mine is the 300 (no „W“) and WB3A. Raw data is „20 cb 00 05“. By that, I‘m using VScotHO1_4, according to this list:
https://github.com/philippoo66/ViessData21/blob/master/DataPoints.txt

KLelong, 2025-04-28:
Sorry for the late reply.
Can you say why you assume to use that VScotHO1_4 ? I do find "20 cb" in the list. But then you have "00 05" which I don't find in the list of "20 cb" lines.
And I don't really understand the "sysHardware/SoftwareIndexIdent"-texts. Are those ranges, so 0100 to 0103 are VScotHO1, 0104 to 0113 are VScotHO_4, etc ? Or what ?