openv/openv #678: VScotHO1/20CB Datenpunkte

aeble, 2025-03-17:
Hi, ich habe eine VScotHO1 mit einer Gasbrennwerttherme hintendran. Die Datei mit den Datenpunkten habe ich angeschaut, allerdings fehlt mir in der Beschreibung eine Angabe der Einheiten (Leistung Brenner - ist das in W, kW, ...). Außerdem suche ich derzeit noch den Datenpunkt für den Wasserspeichersensor, irgendwie finde ich da nix.

Ist hier jemand, der auch seine VScotHO1 ausliest und mit dem ich mich austauschen kann?

BG,
Axel

pDR8iLQ4go, 2025-03-22:
Ich lese die Leistung von der Adresse 0xA38F. Wenn ich den Wert als unsigned integer interpretiere (0…255) und durch 2 teile, habe ich die Leistung in % (0…100). Das, was ich so lese, sieht ganz plausibel aus.

aeble, 2025-03-22:
> Ich lese die Leistung von der Adresse 0xA38F. Wenn ich den Wert (1 Byte) als unsigned integer interpretiere (0…255) und durch 2 teile, habe ich die Leistung in % (0…100). Das, was ich so lese, sieht ganz plausibel aus.

Ah, danke - das ist auch in Prozent... naja, besser als nix. Ich hatte als Teiler 1 drin, dann werde ich das mal ändern auf 2. 

Den Sensor für die Speichertemperatur hast du auch nicht, oder?

pDR8iLQ4go, 2025-03-22:
[image]

Vinshi01, 2025-05-07:
Hello, 

Kannst du evtl deine vcontrold und vita xml senden? Habe auch eine 20CB und bekomme die Betriebsart nicht umgestellt.