openv/openv #681: Vitocal 200-S energy consumption

budisek, 2025-03-27:
Hi

I am looking for any addresses with power consumption values (W/Wh - hourly, daily or total) for energy panel in Home assistant.

Which address of this list here [https://github.com/openv/openv/wiki/Adressen-Wärmepumpen-Vitocal-200‐S-(Bj.---2018)](url)

Thanks

Atomratte, 2025-06-02:
Hi,
I found these addresses:
Address: 0x16A4 (electric power of the compressor in watts)
Address: 0x16A0 (thermal power of the compressor in watts)

Source: https://github.com/openv/openv/issues/650

budisek, 2025-06-03:
Wow, have you tried these? Because it is for different heat pump model. It may not be the same.

Atomratte, 2025-06-03:
I have a Vitocal 200-S with Vitotronic 200 (W01CB), Device ID: 204D.

I tested both of these addresses for one day, and the values appear to be plausible.

Below is a graph from address 0x16A4.
[image]

budisek, 2025-06-03:
Please can you share your address setup? I am pretty lost in this.

Atomratte, 2025-06-03:
I am running an ESP8266 flashed with ESPHome Optolink
(https://github.com/openv/openv/wiki/ESPHome-Optolink).

This is the sensor:

  - platform: optolink
    name: vd_el_leistungsaufnahme
    address: 0x16A4
    bytes: 4
    update_interval: 30s 
    device_class: power
    state_class: measurement
    unit_of_measurement: W
    #Energiebilanz: Aktuelle Leistungsaufnahme Verdichter  (elektrische Leistung)

budisek, 2025-06-03:
Wow you are the best. Its awesome. I will implement these. Thank you very much.

Atomratte, 2025-06-04:
Currently, I am not controlling the heat pump via ESPHome. I may consider doing so in a few months, once the photovoltaic system is installed and I can use the surplus electricity. I do not have the Vitoconnect gateway or the Vitotronic app.

Here are some additional addresses—perhaps they will be helpful to you.
https://community.viessmann.de/viessmann/attachments/viessmann/customers-connectivity/45542/1/vito.xml.pdf

rorso, 2025-08-06:
> Currently, I am not controlling the heat pump via ESPHome. I may consider doing so in a few months, once the photovoltaic system is installed and I can use the surplus electricity. I do not have the Vitoconnect gateway or the Vitotronic app.

I tried some time ago with caution but with good results. Currently I can set the "1x hot-water" and "room temperature value" successfully. The set value can be seen in the device display and  the heat pump does as expected.

However, I found that the reaction of the Vitocal is quite slow when changing the values - in both direction. it may take a couple of minutes(!) before a changed value finally triggers a reaction and likewise when you set it on the device before the change is shown in the value read from Vitocal. No matter how often you read.

So when engaging the "1x hot water" switch and the pump does not start working within a couple of seconds - don't panic. :-)

I do have a 333-G so my addresses might differ:

```
number:
  - platform: optolink
    name: Raumtemperatur HK1 Soll
    unit_of_measurement: °C
    address: 0x2000
    update_interval: 30s
    bytes: 2
    div_ratio: 10
    min_value: 3
    max_value: 37
    step: 1
    mode: box
    icon: "mdi:home-thermometer"
    device_class: temperature

switch:
  - platform: optolink
    name: WW Vorrang
    address: 0x6010
    icon: mdi:heating-coil

  - platform: optolink
    name: WW bei Party
    address: 0x2010
    icon: mdi:water-thermometer

  - platform: optolink
    name: WW Einschaltoptimierung
    address: 0x6009
    icon: mdi:progress-clock

  - platform: optolink
    name: 1x WW
    address: 0xB020
    icon: mdi:water-thermometer
```

I'm going to check the device mode next. These are still untested and I certainly have to watch the "HK Operation Mode", as it manipulates the same address as the "1x hot water" switch. I'm pretty sure this is unneccessary too, as one should not set the value to "manual" at all and what remains is the "1x hot water" that is covered by the switch.

But it would be interesting whether it correctly follows the setting when changed on the device.

```
#select:
#  - platform: optolink
#    name: HK Operation Mode
#    address: 0xb020
#    bytes: 1
#    icon: "mdi:heating-coil"
#    update_interval: 60s
#    map:
#      - "0 -> Normalbetrieb"
##     manueller Betrieb nur nach Rücksprache, KEINE Regelung, kein Zeitprogramm.
#      - "1 -> manueller Betrieb"
#      - "2 -> 1x Warmwasser"

  - platform: optolink
    name: Operation Mode
    address: 0xb000
    bytes: 1
    icon: "mdi:heat-pump-outline"
    update_interval: 60s
    map:
      - "0 -> Abschaltbetrieb"
      - "1 -> nur Warmwasser"
      - "2 -> Heizen u. WW"
```

rorso, 2025-08-06:
Btw, has anyone found a **total energy counter (kW/h)**?

actual power draw is interesting but so would be the total power consumption of the device in the energy dashboard. There you need a "total increasing" counter. Manually integrating the kW is unreliable as you can't guarantee that indeed **every** value gets counted.

Bojkas1985, 2026-08-15:
@rorso I have a Vitocal 242-S. If you want advice on setting it up, I can help you. All you need is a wire to connect it to F11 and have x3.8 and x3.9 connected by wire, which should already be in the factory. The F11 comes with a dew point sensor as standard But I refuse to pay 1000 euros for it to be operational. I bought a relay with a dry contact for switching F11. I monitor the dew point in 5 rooms and on the heating distributions via home assistant. My own solution is more solid and safer, I think. 😁 The wire to connect F11 is a temporary solution and I'll be installing the relay there today. Then just activate the cooling in the service menu for 1 second and voila, it works like a charm. 😁
You can get to service menu 1 by holding down OK and the three menu lines for 5 seconds. Then you can find service, coding, cooling through the menu. You can set everything there. The problem is that it will give you a CA error, which is the dew point sensor, after add wire, it's works.

Bojkas1985, 2026-08-15:
[image]

[image]

rorso, 2026-08-16:
> I have a Vitocal 242-S. If you want advice on setting it up, I can help you. All you need is a wire to connect it to F11 and have x3.8 and x3.9 connected by wire, which should already be in the factory.

Thanks, but I have a 333G that was installed 2010 and simply lacks the necessary pipes, valves and pumps to reverse the thermal circuit. Even if I manage to convince the controller to issue "cooling settings", they are going nowhere.

I have to get a plumber to rework the interior before I can even think about cooling functions. Viessmann service stated:

>Since your Vitocal 333-G is equipped with WO1B control, the current version of the NC-Box cannot be used here. The NC-Box required for your system is no longer manufactured and is no longer available. In that case, the entire system must be rebuilt on-site using the current box, the NC circuit board (7832883), and the appropriate switching valves and circulation pumps.

That doesn't sound cheap, but so far this is not even the biggest problem. I was unsuccessful to convince ANY plumber to at least reply to my request for a quote for more than a month now. :-(

So don't expect any cooling tips from my side any time soon.