openv/openv #688: sensor.SENSOR_SCHEMA deprecated – bitte auf sensor.sensor_schema(...) umstellen

Cupra85, 2025-07-20:
Hallo 👋,

beim Kompilieren mit aktueller ESPHome-Version erhalte ich folgende Warnung:

WARNING Using `sensor.SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. 
Please use `sensor.sensor_schema(...)` instead. 
If you are seeing this, report an issue to the external_component author and ask them to update it.
https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/.
Component using this schema: vitoconnect

Könntest du das bitte im Code anpassen, damit die Komponente auch in Zukunft mit ESPHome >=2025.11.0 weiterhin funktioniert?

Vielen Dank für das großartige Projekt! 🙏

Trainerr, 2025-09-05:
hab leider auch das selbe Problem. Wäre nett eine aktualisierte version zu erhalten

Folllast, 2025-09-25:
Ich befürchte wir können das abschreiben. wir können nur hoffen, dass alte compilierte Programme weiter laufen, oder einfach nicht updaten...

Cupra85, 2025-09-25:
Ich habe die Repository vom Autor kopiert und geändert, der Fehler ist nun weg. 

Außerdem hab ich die maximale Sensorzahl von 20 auf 40 erhöht. Für ein Wemos D1 mini wird es zu viel sein aber z.B. ein ESP32 C6 oder S3 kommt damit gut zurecht!

Ersetzt einfach in eurer YAML die externe Komponente durch diese, dann läuft alles!

`external_components:
  - source:
      type: git
      url: https://github.com/Cupra85/esphome_vitoconnect
      ref: master
    refresh: 0s`

Werner-G, 2025-10-06:
Was mache ich nur falsch? Ich habe Deine Komponente 1.1 eingefügt, erhalte aber diese Fehlermeldung: 

```
INFO ESPHome 2025.9.3
INFO Reading configuration /config/esphome/heizung.yaml...
INFO Updating https://github.com/Cupra85/esphome_vitoconnect@master
INFO Unable to import component optolink: No module named 'esphome.components.optolink'
Failed config

optolink: [source /config/esphome/heizung.yaml:42]
  
  Component not found: optolink.
  protocol: P300
```